FreedomBen/rtl8188ce-linux-driver #59: Kernel Panic on Linux Mint 17.1

xinis, 2015-01-04:
Hi Freedom Ben, 

Im really a rookie on the Linux System. So far , two days experiance :)
I have currently on one my ASUS R704V the Linux Mint 17.1 on it.
Unfortunately the Asus has the Realtek RTL8188CE as a Wireless card onboard. 
I used your full automatic script and as well the manaully way and all of them generate a KERNEL Panic.
Do you have a good advise for me ?
Regards
Sini

FreedomBen, 2015-01-05:
Welcome to Linux!  I hope you don't let this sour your experience too much ;-)  I *love* Linux but it did take a bit to get there, but I'm so glad it did!  I can't use anything else anymore and be happy :-)

Can you post the output of `uname -a` and `lsb_release -d`?  Also do you remember which branch the script told you to pick?

wintergit, 2015-01-05:
Hello, I'm having the same issue on Linux Mint 17.1 with Realtek RTL8188CE.

uname -a:
Linux laptop 3.13.0-37-generic #64-Ubuntu SMP Mon Sep 22 21:28:38 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux

lsb_release -d:
Description: Linux Mint 17.1 Rebecca

branch:
mint-17

Thanks

wintergit, 2015-01-05:
The kernel panic occurs on step 6: sudo modprobe rtl8192ce. Any ideas on what the issue might be and how I can help? Thank you

FreedomBen, 2015-01-05:
Ok, I think I may have got it.  @wintergit Would you be willing to pull down the latest and give it another try?  `git pull --rebase`

FreedomBen, 2015-01-05:
There is now a `mint-17.1` branch by the way, so make sure it switches you properly

wintergit, 2015-01-05:
Works perfectly now, thanks for your help!

FreedomBen, 2015-01-05:
Awesome, thanks!

closed 2015-01-05 by FreedomBen

xinis, 2015-01-05:
Hi Ben 

so - I didnt it again and :) - the Automatic Installation and all runs smooth during the installation and the reboot. No Kernel Panic anymore.
Also I check the status with "am_I_using..etc" and I have the new drivers on it.
But Ben .. Sorry to bother you again. I have now the drivers on my Linux Mint 17.1 but I still do not see my Wifi/Wlan.
How can I start the "driver" for Wlan ?
Sorry for my "rookie" questions ..

xinis, 2015-01-06:
HI Ben , I was lucky and found out the issue ( in a German Forum )
I have a Asus R704V Laptop and the strange is that the network is "blocked" 
rfkill List 
0: phy0: Wireless LAN
Soft blocked: No
Hard blocked: Yes

So - I have created the following file 
echo "options asus_nb_wmi wapf=1" | sudo tee /etc/modprobe.d/asus.conf
Now - your PERFECT DRIVER runs smoothhhh
Thanks Ben and have a great day
Regards from Germany
Sini

FreedomBen, 2015-01-06:
Awesome @xinis , thanks for the follow-up!

commit 8a5b864a (2015-01-05): Mint-17 - switch to _ieee80211_is_robust_mgmt_frame( hdr )

Fixes #59