FreedomBen/rtl8188ce-linux-driver #69: RTL8188CE No modulation settings; no 802.11n.

agathokles, 2015-02-22:
Newbie here on Ubuntu etc., so please be gentle; on 14.04 LTS.
My laptop has a Realtek 8188CE combo PCI-E NIC; as you...very probably know this NIC is supposed to support all three of the b/g/n.
But it seems that on Ubuntu the n part is missing, is not used.
On the same machine on Windows 7 on the other hand, the n part is also used connecting nominally in 150Mbps and achieving more or less thrice or four times the actual rate (actual measurement on personal lan by transfering huge files) I get on Ubuntu be it with the 14.04 default driver or with this one which I've installed trying to solve the problem; I can only get at most a nominal 54Mbps rate (had inter alia also to set the rate manually cause before that the automatic method was casuing considerably lower rates  and unstable ones, despite available signal power), i.e. around 2.5MBps (again, actual measurement).
Moreover  I simply cannot list or change the modulation settings/options.
May I ask whether this is a bug or a feature?
Or is it just me?
Thanx in advance.

```lspci |grep -i network
02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter (rev 01)```

```modinfo rtl8192ce
filename:       /lib/modules/3.13.0-45-generic/kernel/drivers/net/wireless/rtlwifi/rtl8192ce/rtl8192ce.ko
firmware:       rtlwifi/rtl8192cfwU_B.bin
firmware:       rtlwifi/rtl8192cfwU.bin
firmware:       rtlwifi/rtl8192cfw.bin
description:    Realtek 8192C/8188C 802.11n PCI wireless
license:        GPL
author:         Benjamin Porter	<BenjaminPorter86@gmail.com>
author:         Larry Finger	<Larry.Finger@lwfinger.net>
author:         Realtek WlanFAE	<wlanfae@realtek.com>
author:         lizhaoming	<chaoming_li@realsil.com.cn>
srcversion:     53EBB785CF741A05F34B0A0
alias:          pci:v000010ECd00008176sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008177sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008178sv*sd*bc*sc*i*
alias:          pci:v000010ECd00008191sv*sd*bc*sc*i*
depends:        rtlwifi,rtl8192c-common,mac80211
vermagic:       3.13.0-45-generic SMP mod_unload modversions 
parm:           swenc:Set to 1 for software crypto (default 0)
 (bool)
parm:           ips:Set to 0 to not use link power save (default 1)
 (bool)
parm:           swlps:Set to 1 to use SW control power save (default 0)
 (bool)
parm:           fwlps:Set to 1 to use FW control power save (default 1)
 (bool)
parm:           debug:Set debug level (0-5) (default 0) (int)```

```lshw
...
*-network
                description: Wireless interface
                product: RTL8188CE 802.11b/g/n WiFi Adapter
                vendor: Realtek Semiconductor Co., Ltd.
                physical id: 0
                bus info: pci@0000:02:00.0
                logical name: wlan0
                version: 01
                serial: C8:3A:**:**:**:**
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
                configuration: broadcast=yes driver=rtl8192ce driverversion=3.13.0-45-generic firmware=N/A ip=192.168.1.2 latency=0 link=yes multicast=yes wireless=IEEE 802.11bgn
                resources: irq:18 ioport:e000(size=256) memory:f7d00000-f7d03fff
...```

```iwconfig
...
wlan0     IEEE 802.11bgn  ESSID:"*******************"  
          Mode:Managed  Frequency:2.472 GHz  Access Point: C8:3A:**:**:**:**   
          Bit Rate=54 Mb/s   Tx-Power=33 dBm   
          Retry  long limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:off
          Link Quality=70/70  Signal level=-38 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:82   Missed beacon:0
...```

```iwlist scan
wlan0     Scan completed :
          Cell 01 - Address: C8:3A:**:**:**:**
                    Channel:13
                    Frequency:2.472 GHz (Channel 13)
                    Quality=70/70  Signal level=-29 dBm  
                    Encryption key:on
                    ESSID:"************"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=00000003fea57fd4
                    Extra: Last beacon: 47584ms ago
                    IE: Unknown: 00084153454D41544F4E
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 03010D
                    IE: Unknown: 2A0100
                    IE: Unknown: 2F0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 32040C121860
                    IE: Unknown: 2D1AFE181BFFFF000001000000000000000000000000000000000000
                    IE: Unknown: 3D160D0F0000000000000000000000000000000000000000
                    IE: Unknown: DD090010180201F0280000```

```iwlist chan
....
wlan0     13 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Current Frequency:2.472 GHz (Channel 13)
....```

```iwlist wlan0 event
wlan0     Wireless Events supported :
          0x8B04 : Set Frequency/Channel (kernel generated)
          0x8B06 : Set Mode (kernel generated)
          0x8B15 : New Access Point/Cell address - roaming
          0x8B19 : Scan request completed
          0x8B1A : Set ESSID (kernel generated)
          0x8B2A : Set Encoding (kernel generated)

wlan0     Scanning capabilities :
		- ESSID```

agathokles, 2015-02-23:
A seconday if not tertiary detail but anyway, just to be clear, I've copied -and partially hidden- the values of the serial and the macs wrongly.
Oh and I'm running  the 64b version of the 14.04LTS.

agathokles, 2015-03-16:
Will there be no answer? I've been waiting for weeks...
Not even a comment? Not even something like dude "you're wrong about this because" or "you're doing this wrong because" or "this is not the place for this because"...?!?

OK let me rephrase:

Apart from the lack of modulation options and the impossibility of setting to 802.11n, IT IS ALSO IMPOSSIBLE on my computer, using this or the standard version of the driver, to set the tx power to more than 20dBm.
There.
Could now please someone help?! 
Or say something at least?!? ;-)

FreedomBen, 2015-03-16:
Sorry, I've been insanely busy lately.  I'll try to reply with something soon

agathokles, 2015-03-17:
A note:
"Tx-Power=33 dBm" above seems to be a fluke; 99.999% of the time (I can't even remember managing to set the value this high) when I try to set tx power to above 20, nothing happens, the value remains 20dBm, as at least reported by iwconfig.

FreedomBen, 2015-03-22:
There are no modulation settings that I'm aware of.  However, you should be able to go above 20 dBm so long as your CRDA settings don't restrict it.  Have you tried setting CRDA to BO?

agathokles, 2015-03-22:
With your reply ( i.e. no modulation change possible, settings available) and after the things I finally found out yesterday, the mystery and the problem is probably solved, though not without its quirks and with full certainty.

It seems that Windows can/do connect to an AC in 802.11n mode even when WMM/QoS is disabled in the AC while Ubuntu can't or doesn't (as seen in the dmesg output; see for example [here](http://ubuntuforums.org/showthread.php?t=2244010)). 

But this is prima facie invisible to the user using this NIC, OS and driver, at least in my PC or PC-AC configuration.
That is:
1. Enabling the relevant WMM/QoS settings in the AC caused the rate, as seen in e.g. a Nautilus copy/paste, to reach 802.11n levels. This got reflected in the AC connection status showing a 40MHz wide channel present. I THINK THAT THIS HAS TO BE EXPLICITLY STATED SOMEWHERE IN LINUX/UBUNTU/... and relevant sites/pages,  whatever the NIC and/or the driver thereof. Please add it for the sake of other people to the README.md.
2. On the other hand, this is not reflected in the rate shown in the Connection Information of the Ubuntu Network Manager nor .e.g in the output of iwconfig. Moreover, trying to set the XXX in "sudo iwconfig INTERFACE rate XXX" to anything more than 54M causes the NIC and/or the driver to go crazy and revert to a 1M rate. THIS MAKES THINGS UNCERTAIN AND TROUBLING AND SEEMS LIKE A BUG; again at least with my system/network configuration, that is.

PS On the txpower>20dBm issue I have to say 
1. MEA CULPA, APOLOGIES!
2. I may try it out in the future; I'm presently satisfied with having solved or probably solved the n mode problem.
3. More details: I hadn't tried to change the relevant regional settings. Had forgotten about this needed change -although unlike the WMM/QoS thing, instructions about it are readily available indeed-  whilst being overwhelmed by the n mode problem. 
I only first did a "sudo iw reg set XX", setting XX to my country's code after seeing the output of dmesg setting XX to CN when connecting a USB NIC I got my hands onto during the last days, to the laptop and whilst looking inter alia for a solution to n mode problem through comparison of the NICs and the settings thereof (btw: no modulation change possible for that NIC either). Though, and this is going even more offtopic, for the life of me, after reading [such stuff](https://wireless.wiki.kernel.org/en/developers/regulatory/crda)
"In order to achieve this devices always respect their programmed regulatory domain and a country code selection will only enhance regulatory restrictions"
I can't find where the "correct" wpa_supplicant file is located in the system so that the proper domain gets set ONCE at boot time without - the fear of - piling up regulatory restrictions...

jcdenton2k, 2015-05-12:
I want to thank the OP a ton for figuring this out.  You are a lifesaver!  :+1:

agathokles, 2015-05-12:
@jcdenton2k and whomever else this might prove helpful to:

I've finally managed to also(?) change the nominal rate to show 802.11n rates. 
Solution?
Use **iw** instead of **iwconfig**...
Go figure...

From iw help:
```dev <devname> set bitrates [legacy-<2.4|5> <legacy rate in Mbps>*] [mcs-<2.4|5> <MCS index>*]
		Sets up the specified rate masks.
		Not passing any arguments would clear the existing mask (if any)```
The *2.4* or *5* part (for god's sake, why do we have to decipher text that is supposed to help us find how to do things?) refers to the 2.4GHz or 5GHz spectrum (therefore use *2.4*).
For MCS index values see [this](http://en.wikipedia.org/wiki/IEEE_802.11n-2009#Data_rates).

Using *legacy-2.4* - at least in my case - does not seem to work for - among other things -values greater than 54; error message:
```command failed: Invalid argument (-22)```

So use something like this:
```iw dev wlan0 set bitrates mcs-2.4 7```

FreedomBen, 2015-05-13:
Wow, very interesting and helpful.  Thanks!

duby229, 2015-06-01:
So, I'm pretty sure this describes the same problem I was having. I just happened to come back to this site to check it out and come across this thread. I'm pretty sure you just described the same problem I'm having. Except in my case I'm still connecting through a 802.11g AP. I know it needs updated, but it's the office access point, until then I just have to deal with it.

I don't have the ability to gain access to the AP interface, so I don't know if it has any configurable options. 

Also I'm using wicd for a GUI, which doesn't utilize iw. I'm going to try shutting down wicd and then reconnect with iw and see if it performs any better.

agathokles, 2015-06-05:
duby, provided that I've understood you, that you're facing the same problem,
try something like this
```
dmesg | grep -i WMM/QoS
```
If the command returns, as mentioned in the relevant linked-to discussion,
something like this
```
[48169.134778] iwlwifi 0000:02:00.0 wlan0: disabling HT as WMM/QoS is not supported by the AP
[48169.134781] iwlwifi 0000:02:00.0 wlan0: disabling VHT as WMM/QoS is not supported by the AP
```
then afaict it's the AP settings' fault (provided it's indeed a 802.11n AP) as explained above, along with Ubuntu/Linux following strictly the standard and being thus, let's say, not that user-friendly.
However if the command does return such an output then I don't think changing the program used to configure the NIC/WLAN  would make any difference.
But I guess OK, it wouldn't hurt to try...

jcdenton2k, 2015-11-16:
Quick Summary of this thread for those who want a TLDR:

-For Linux distros, you should enable WMM/QoS on the router/access point in order to get maximum possible speeds (Wireless N) from the network.

-Use the terminal command: dmesg | grep -i WMM/QoS.
--If you get any output then it is a router/access point issue and not an issue on your end or with the driver.

FreedomBen, 2015-11-17:
awesome, thanks @jcdenton2k :+1: