FreedomBen/rtl8188ce-linux-driver #71: Ubuntu Kernel v3.18.10 is not supported

jcdenton2k, 2015-03-27:
Building fails on v3.18.10 and v3.19.* versions of the Ubuntu kernel.
I was especially surprised by the 3.18.10-031810-generic failure.

I did have success with the v3.18.5 version:
http://kernel.ubuntu.com/~kernel-ppa/mainline/v3.18.5-vivid/

So I would like for you to add in support for v3.18.10 at least as well as hopefuly add in support for v3.19.* versions and the upcoming v4.0 once it is released.

Thanks again :)

FreedomBen, 2015-03-28:
thanks for letting me know.  I'll try to get it working

jcdenton2k, 2015-03-28:
I appreciate the response, especially so quickly!
If you need me to do so, I can test it with every version in between 3.18.5 and 3.18.10 if you need me to narrow down the 'build fail' reason.
It would take me a bit of time to test each one but I could dedicate a day to doing so if it'll help us get support done faster.

Makes me wish I had some programming knowledge to do this technical wizardry that you do ^_^
Thank you again and take care!

lowbassman, 2015-04-03:
Attempting to build with a 3.18.9 kernel on a USBArmory device: http://www.inversepath.com/usbarmory.html

  CC [M]  /home/m/local/downloads/rtl8188ce-linux-driver/pci.o
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_parse_configuration’:
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c:433:2: error: implicit declaration of function ‘pcie_capability_read_word’ [-Werror=implicit-function-declaration]
  pcie_capability_read_word( pdev, PCI_EXP_LNKCTL, &linkctrl_reg );
  ^
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_intr_mode_msi’:
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c:2108:2: error: implicit declaration of function ‘pci_enable_msi’ [-Werror=implicit-function-declaration]
  ret = pci_enable_msi( rtlpci->pdev );
  ^
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c:2115:3: error: implicit declaration of function ‘pci_disable_msi’ [-Werror=implicit-function-declaration]
   pci_disable_msi( rtlpci->pdev );
   ^
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_get_amd_l1_patch’:
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c:355:17: warning: ‘offset_e0’ is used uninitialized in this function [-Wuninitialized]
  if ( offset_e0 == 0xA0 ) {
                 ^
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_get_linkcontrol_field’:
/home/m/local/downloads/rtl8188ce-linux-driver/pci.c:420:46: warning: ‘linkctrl_reg’ is used uninitialized in this function [-Wuninitialized]
  pcipriv->ndis_adapter.pcibridge_linkctrlreg = linkctrl_reg;

jcdenton2k, 2015-04-16:
Just an update.  Requesting compatability with 3.18.10 (and later) as well as the 3.19.x branch and the newly-released 4.x mainline branch :)

I'm curious as I haven't had any further issues with the default wifi drivers.  Perhaps they included some of your code automatically in the newer kernels?

I'll have to do some more testing.  Thank you again for everything, FreedomBen :)

FreedomBen, 2015-04-19:
Ok I just made a `generic-4.0.x` branch so that should work on most distros.  Ubuntu does have a nasty habit of hacking and slashing their kernel :confounded: , which makes it really hard to support all of their kernels.  I've basically just started supporting the "official" ubuntu kernels to avoid a tremendous amount of work.

Are you committed to the 3.18 ubuntu-flavored kernel?

jcdenton2k, 2015-04-19:
I'm actually on the 4.0.x branch right now.  If the 3.18.6+ and 3.19.x branches of the Ubuntu kernel take too much time to support then don't worry about em.

I'll test out your 4.0.x branch in a day or two and update here with the results.  Not sure how to save logs/debug output in case something goes south.

jcdenton2k, 2015-05-12:
Please close this issue.  I can confirm that the 4.0.2 'Wily' kernel from:
http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.0.2-wily/

Compiles and works with ZERO issues.  So awesome :+1:

Please have my babies and set up an online tip jar (maybe join flattr or something) so that people can throw some money at you for some food/drink of your choice :)

closed 2015-05-12 by jcdenton2k

FreedomBen, 2015-05-13:
haha, glad it worked!