FreedomBen/rtl8188ce-linux-driver #74: Kernel Panic on Ubuntu 12.04.5

HenryJacques, 2015-06-08:
Hello,

Using Ubuntu 12.04.5 64bits (with kernel 3.13), I got a kernel panic (see picture)
[image]

I used the manual installation using the generic-3.13.x branch. It compiled successfully and installed it as well. But at the reboot I got a kernel panic....

Cheers

FreedomBen, 2015-06-08:
afaik, if you're using the ubuntu-flavored 3.13.x that ships with ubuntu, you have to use the `ubuntu-12.04` branch or you do get a kernel panic.  They call an old/incompatible version of a certain function.   Is there a reason you need the generic branch?

HenryJacques, 2015-06-09:
Thanks for coming back to me.
I used the generic branch because I thought that the ubuntu-12.04 branch was restricted to the base kernel (3.2.x). Since I am using the linux-generic-lts-trusty kernel (3.13.X) I used the generic 3.13.X
Using the 12.04 branch I got the following error:

 verify_branch.sh ] ; \
        then \
            ./verify_branch.sh ; \
        fi;
Verifying a sane branch for your kernel version...
No (current branch rtl8188ce-linux-driver-ubuntu-12.04)
-e 
You don't appear to be in a Git checkout.
This means branch information is not available.
You can still proceed to build, but you might be using unstable code.
Would you like me to try and get a git checkout for you? (Y/N): n
Continue with the build? (Y/N): y
make -C /lib/modules/3.13.0-53-generic/build M=/tmp/rtl8188ce-linux-driver-ubuntu-12.04 modules
make[1]: entrant dans le répertoire « /usr/src/linux-headers-3.13.0-53-generic »
  CC [M]  /tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.o
/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.c: In function ‘rtl_action_proc’:
/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.c:892:32: erreur: ‘struct ieee80211_conf’ has no member named ‘channel’
/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.c:893:32: erreur: ‘struct ieee80211_conf’ has no member named ‘channel’
/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.c: In function ‘rtl_send_smps_action’:
/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.c:1458:24: erreur: ‘struct ieee80211_conf’ has no member named ‘channel’
make[2]: *** [/tmp/rtl8188ce-linux-driver-ubuntu-12.04/base.o] Erreur 1
make[1]: *** [_module_/tmp/rtl8188ce-linux-driver-ubuntu-12.04] Erreur 2
make[1]: quittant le répertoire « /usr/src/linux-headers-3.13.0-53-generic »
make: *** [all] Erreur 2

HenryJacques, 2015-06-09:
I tried the generic-14.04 branch. It compiled successfully. (however what is the diffence with the 14.04-legacy branch?)
However when I try to use the module:

FATAL: Error inserting rtl8192ce (/lib/modules/3.13.0-53-generic/updates/drivers/net/wireless/rtlwifi/rtl8192ce/rtl8192ce.ko): Invalid argument

I also tried without the arguments in /etc/modprobe.d/modprobe.conf

:(

HenryJacques, 2015-06-09:
Oh oh, I used the 14.04-legacy and it seems to work great! No more disconnection and it seems much more stable! (if you want to may update your readme file)
I will see on the long run but it seems promising!
Thank you very much!

HenryJacques, 2015-06-09:
Me, again...
I spoke too quickly, I also have the "Invalid argument" error with the 14.04-legacy branch. It seems I did too much attemps on my workstation. Using a fresh install I still have the error...

FreedomBen, 2015-06-09:
> I thought that the ubuntu-12.04 branch was restricted to the base kernel (3.2.x).

Ah yeah, you're totally right.  Sorry for the confusion.

So are you running an official Ubuntu kernel 3.13 then?  If so then the `ubuntu-14.04` would be the most likely to not panic.  I'll take a look at the diff and see if ubuntu has done some shifting

HenryJacques, 2015-06-10:
Yeah I'm running the official ubuntu kernel 3.13.
the ubuntu-14.04 branch don't cause any kernel panic but I'm unable to load the module due to the "Invalid argument" error...
Will updating the linux-firmware package help?

Thanks!

HenryJacques, 2015-06-10:
Update: I did a fresh install and the module load successfully (using the ubuntu-14.04 branch).
However, the connexion lasts only a few seconds before I got disconnected:

[mer. juin 10 09:51:47 2015] wlan0: deauthenticating from c8:cb:b8:fd:3c:11 by local choice (reason=3)
[mer. juin 10 09:51:47 2015] cfg80211: Calling CRDA to update world regulatory domain
[mer. juin 10 09:51:47 2015] cfg80211: World regulatory domain updated:
[mer. juin 10 09:51:47 2015] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp)
[mer. juin 10 09:51:47 2015] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
[mer. juin 10 09:51:47 2015] cfg80211:   (2457000 KHz - 2482000 KHz @ 20000 KHz), (300 mBi, 2000 mBm)
[mer. juin 10 09:51:47 2015] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (300 mBi, 2000 mBm)
[mer. juin 10 09:51:47 2015] cfg80211:   (5170000 KHz - 5250000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
[mer. juin 10 09:51:47 2015] cfg80211:   (5735000 KHz - 5835000 KHz @ 40000 KHz), (300 mBi, 2000 mBm)
[mer. juin 10 09:51:47 2015] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

HenryJacques, 2015-06-10:
Finally, I'm getting somewhere, it works!

For your Info:
- Realtek RTL8188CE (on Lenovo L430)
- Ubuntu 12.04.5 (with 3.13 kernel)
- I use the ubuntu-14.04 branch with the modprobe options
- I *must* fix the data rates to 24M (as described in your readme)
- I *must* use the stock linux-firmware package for 12.04 (and not use a more recent one)

Many thanks for your feedback and sorry for the numerous posts here ;)

Bye

closed 2015-06-10 by HenryJacques

FreedomBen, 2015-06-10:
Awesome!  I'm glad you got it working.  Thanks for the follow-up info as well.  I'm sure that will help people in the future.