FreedomBen/rtl8188ce-linux-driver #75: after installing this driver it shows that it still is running the stock driver

feardax, 2015-06-13:
i am using kali 1.0.8 (kernel 3.14) and my wifi adapter chipset is rtl8192cu because i was problems with other drivers i installed this but now  when  i do 

./am_i_using_this_driver.sh 

i get 

[*] You are running the new rtlwifi
[*] Not running the new rtl8192cu (running stock driver)

and now wireless networks are not showing up. 
could you help me out here thanks in advance.

MRebhan, 2015-06-27:
Confirmed on Ubuntu 15.04 with rtl8192cu driver.

etem, 2016-09-11:
Confirmed on Ubuntu 16.04 (kernel 4.7.2).
After successful installation and reboot, i still get : 

[*] Not running the new rtlwifi (running stock driver)
[*] Not running the new rtl8192c_common (running stock driver)
[*] Not running the new rtl8192ce (running stock driver)

julianofischer, 2016-11-14:
Confirmed on Ubuntu 16.04 (kernel 4.4.0).

    [*] Not running the new rtlwifi (running stock driver)
    [*] Not running the new rtl8192c_common (running stock driver)
    [*] Not running the new rtl8192ce (running stock driver)

TrackerSB, 2017-04-07:
Confirmed on Ubuntu 16.10 (kernel 4.8.0)
    `[*] You are running the new rtlwifi`
    `[*] Not running the new rtl8192cu (running stock driver)`