FreedomBen/rtl8188ce-linux-driver #76: Can't connect to Wireless N network.

x0a, 2015-06-23:
Iwconfig shows my bit rate as perpetually stuck at 18 megabits/s. I also can't connect to N-only networks.

x0a, 2015-06-24:
I've figured it out! The problem isn't the driver, it's the router settings. I hadn't suspected the router because the stock driver connected okay, but that is actually incorrect behavior. Wireless N requires WMM support to be enabled in order to allow speeds greater than 54 mbps. WMM and QoS both contribute to the high throughput of Wireless N, thus a lot of devices will refuse to connect to a network under Wireless N if WMM is disabled.  So this is completely normal behavior.

closed 2015-06-24 by x0a

FreedomBen, 2015-06-24:
Awesome, thank you!