FreedomBen/rtl8188ce-linux-driver #77: Cannot compile on raspberry pi

MRebhan, 2015-06-25:
This does not compile on raspberry pi because it has no PCI

It always fails with this message:

pi@raspberrypi ~/rtl8188ce-linux-driver (git)-[generic-4.0.x] % make
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
Yes
make  -C /opt/kernel/build M=/home/pi/rtl8188ce-linux-driver modules
make[1]: Entering directory '/opt/kernel/build'
  CC [M]  /home/pi/rtl8188ce-linux-driver/base.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/usb.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/rc.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/debug.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/regd.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/efuse.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/cam.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/ps.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/core.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/stats.o
  CC [M]  /home/pi/rtl8188ce-linux-driver/pci.o
/home/pi/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_parse_configuration’:
/home/pi/rtl8188ce-linux-driver/pci.c:433:2: error: implicit declaration of function ‘pcie_capability_read_word’ [-Werror=implicit-function-declaration]
/home/pi/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_intr_mode_msi’:
/home/pi/rtl8188ce-linux-driver/pci.c:2133:2: error: implicit declaration of function ‘pci_enable_msi’ [-Werror=implicit-function-declaration]
/home/pi/rtl8188ce-linux-driver/pci.c:2140:3: error: implicit declaration of function ‘pci_disable_msi’ [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
/opt/kernel/source/scripts/Makefile.build:258: recipe for target '/home/pi/rtl8188ce-linux-driver/pci.o' failed
make[4]: *** [/home/pi/rtl8188ce-linux-driver/pci.o] Error 1
/opt/kernel/source/Makefile:1390: recipe for target '_module_/home/pi/rtl8188ce-linux-driver' failed
make[3]: *** [_module_/home/pi/rtl8188ce-linux-driver] Error 2
Makefile:145: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:24: recipe for target '__sub-make' failed
make[1]: *** [__sub-make] Error 2
make[1]: Leaving directory '/opt/kernel/build'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2
make  145.05s user 7.81s system 97% cpu 2:36.37 total
2 pi@raspberrypi ~/rtl8188ce-linux-driver (git)-[generic-4.0.x] %

FreedomBen, 2015-07-15:
I'll try to get to this, but I can't make any promises due to lack of time :-(

MRebhan, 2015-07-18:
Ok, let me know if you fixed it!
Thanks

pier22, 2017-09-19:
Hello,

I have had the same problem with my orange development board Orange pi.
The following error occurred while compiling..
...
  CC [M]  /home/pier2/rtl8188ce-linux-driver/core.o
  CC [M]  /home/pier2/rtl8188ce-linux-driver/stats.o
  CC [M]  /home/pier2/rtl8188ce-linux-driver/pci.o
/home/pier2/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_parse_configuration’:
/home/pier2/rtl8188ce-linux-driver/pci.c:427:2: error: implicit declaration of function ‘pcie_capability_read_word’ [-Werror=implicit-function-declaration]
  pcie_capability_read_word( pdev, PCI_EXP_LNKCTL, &linkctrl_reg );
  ^
/home/pier2/rtl8188ce-linux-driver/pci.c: In function ‘rtl_pci_intr_mode_msi’:
/home/pier2/rtl8188ce-linux-driver/pci.c:2120:8: error: implicit declaration of function ‘pci_enable_msi’ [-Werror=implicit-function-declaration]
  ret = pci_enable_msi( rtlpci->pdev );
        ^
/home/pier2/rtl8188ce-linux-driver/pci.c:2127:3: error: implicit declaration of function ‘pci_disable_msi’ [-Werror=implicit-function-declaration]
   pci_disable_msi( rtlpci->pdev );
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:294: recipe for target '/home/pier2/rtl8188ce-linux-driver/pci.o' failed
make[2]: *** [/home/pier2/rtl8188ce-linux-driver/pci.o] Error 1
Makefile:1492: recipe for target '_module_/home/pier2/rtl8188ce-linux-driver' failed
make[1]: *** [_module_/home/pier2/rtl8188ce-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.11.1-sun50iw2'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2

I would like to tell if I already solve the problem ..?, Or if it did not, maybe it would be simple to prevent the compile for the pio port, since it is not present on both development boards, Raspberry and Orange ..

Thank you