FreedomBen/rtl8188ce-linux-driver #78: implicit declaration of function ‘rtl_hal_pwrseqcmdparsing’

unallocatedanonymous, 2015-07-02:
Unable to compile on Fedora 21, 3.18 kernel due to the following error:
implicit declaration of function ‘rtl_hal_pwrseqcmdparsing’

Found this patch which solved the problem:
http://www.spinics.net/lists/linux-wireless/msg134699.html

Sergui2015, 2015-07-25:
How do you apply this solution? What do you need to modify?

FreedomBen, 2015-07-27:
@Sergui2015 :  Open up the file `rtl8188ee/hw.c` and add this include:

    #include "../pwrseqcmd.h"

and you should be golden.

ulrith, 2015-09-10:
I have this problem on Ubuntu 15.04.

When run install.sh I've got the error:

rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/rtl8188ee/hw.c:860:2: error: implicit declaration of function ‘rtl_hal_pwrseqcmdparsing’ [-Werror=implicit-function-declaration]
  if ( !rtl_hal_pwrseqcmdparsing( rtlpriv, PWR_CUT_ALL_MSK,
  ^
cc1: some warnings being treated as errors

But I was unable to fix this by adding the line
#include "../pwrseqcmd.h"
to file rtl8188ce-linux-driver/rtl8188ee/hw.c

because right after downloading the zip I already HAVE this entry in this file.

Then after I run install.sh I've got the error listed above and the entry disappeared from file hw.c.

If I add it again, I can't ran install.sh again:

[*] Now let's compile the driver from source and copy the files to the right directories
make: *** No targets specified and no makefile found.  Stop.
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)

Please help

ulrith, 2015-09-10:
Solved by commenting lines 43-46 in install.sh

meneal, 2016-01-08:
The latest comment on this is unfortunately incorrect.  Commenting out those lines just ignores the fact that there is an error.  The correct fix for the 15.04 branch is to add change the entire preamble of hw.c to the following:

```
#include "../wifi.h"
#include "../efuse.h"
#include "../base.h"
#include "../regd.h"
#include "../cam.h"
#include "../ps.h"
#include "../pci.h"
#include "../pwrseqcmd.h"
#include "reg.h"
#include "def.h"
#include "phy.h"
#include "dm.h"
#include "fw.h"
#include "led.h"
#include "hw.h"
#include "pwrseq.h"
```

This is the same as it is in the root branch.  Making the change above of adding in ``` #include "../pwrseqcmd.h"``` is dependent on its position in the #includes as far as I have seen.  I stupidly tried to add it at the end under pwrseq.h and that totally failed.

ulrith, 2016-01-12:
Thank you meneal, I tried to install the fresh version with no luck:

```  MODPOST 1 modules
  CC      /home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8192de/rtl8192de.mod.o
  LD [M]  /home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8192de/rtl8192de.ko
make[2]: Leaving directory '/usr/src/linux-headers-3.19.0-43-generic'
make[1]: Leaving directory '/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8192de'
make[1]: Entering directory '/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee'
make -C /lib/modules/3.19.0-43-generic/build M=/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee modules
make[2]: Entering directory '/usr/src/linux-headers-3.19.0-43-generic'
  CC [M]  /home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee/hw.o
/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee/hw.c: In function '_rtl88ee_init_mac':
/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee/hw.c:860:2: error: implicit declaration of function 'rtl_hal_pwrseqcmdparsing' [-Werror=implicit-function-declaration]
  if ( !rtl_hal_pwrseqcmdparsing( rtlpriv, PWR_CUT_ALL_MSK,
  ^
cc1: some warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee/hw.o' failed
make[3]: *** [/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee/hw.o] Error 1
Makefile:1395: recipe for target '_module_/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee' failed
make[2]: *** [_module_/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-3.19.0-43-generic'
Makefile:33: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/ulrith/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ce-linux-driver/rtl8188ee'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)
```

ulrith, 2016-01-12:
Thank you meneal, your fix is okay, should it be added to the repository?...

meneal, 2016-01-12:
Yeah I'd think that someone should do a pull request.  More likely than not
it should be me.  Whenever I get a chance I will try to do so.

FreedomBen, 2016-01-12:
@meneal that would be great.  If someone can get a pull request put together, I'll merge it.