FreedomBen/rtl8188ce-linux-driver #79: Ubuntu 14.04 with 3.16.0-45-generic kernel

taniadaniela, 2015-08-06:
I'm using Elementary OS Freya (based on Ubuntu 14.04) and I've tried to install this driver but I'm having this message:
Recommended branch is ubuntu-14.10 based on your kernel version (3.16.0-45-generic)
Should I switch it to ubuntu-14.10 for you?  (y/n):

It's safe to use this branch?

(I already installed the driver with elementary luna based on ubuntu 12.04 but it didnt resolve my dropouts problem, I want to try now with this newer version of the OS)

FreedomBen, 2015-08-10:
Sorry for the delay.  Are you using an ubuntu-flavored kernel or a generic one?

taniadaniela, 2015-08-10:
I think is generic, uname -a gives me Linux XXXXX 3.16.0-45-generic #60~14.04.1-Ubuntu

FreedomBen, 2015-08-11:
ok that looks like an Ubuntu-flavored kernel.  You should be ok to take the recommendation and use `ubuntu-14.10` branch

closed 2016-01-09 by FreedomBen