FreedomBen/rtl8188ce-linux-driver #80: Ubuntu 15.04 won't make

sirEgghead, 2015-08-11:
Trying the manual install and the install script.  Neither one make it through the compile phase.

scripts/Makefile.build:257: recipe for target '/home/username/rtl8188ce-linux-driver/rtl8188ce/hw.o' failed

vgallur, 2015-10-11:
Same here using the ubuntu 15.04 branch. However, switching to generic-3.19.x branch worked, with kernel 3.19.0-30.

FreedomBen, 2015-10-11:
Oh bummer.  Will look at this

FreedomBen, 2015-10-17:
I finally got a chance to take a look and it doesn't look like a quick fix.  Will try to revisit when I get some time

jaredharley, 2015-12-19:
Had the same problem on Linux Mint 17.3 Cinnamon 64-bit (3.19.0-32-generic). When I tried building with the Ubuntu 15.04 branch, failed with the following output:

```
make[1]: Entering directory `/home/jared/git/rtl8188ce-linux-driver/rtl8188ee'
make -C /lib/modules/3.19.0-32-generic/build M=/home/jared/git/rtl8188ce-linux-driver/rtl8188ee modules
make[2]: Entering directory `/usr/src/linux-headers-3.19.0-32-generic'
  CC [M]  /home/jared/git/rtl8188ce-linux-driver/rtl8188ee/hw.o
/home/jared/git/rtl8188ce-linux-driver/rtl8188ee/hw.c: In function ‘_rtl88ee_init_mac’:
/home/jared/git/rtl8188ce-linux-driver/rtl8188ee/hw.c:860:2: error: implicit declaration of function ‘rtl_hal_pwrseqcmdparsing’ [-Werror=implicit-function-declaration]
  if ( !rtl_hal_pwrseqcmdparsing( rtlpriv, PWR_CUT_ALL_MSK,
  ^
cc1: some warnings being treated as errors
make[3]: *** [/home/jared/git/rtl8188ce-linux-driver/rtl8188ee/hw.o] Error 1
make[2]: *** [_module_/home/jared/git/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.19.0-32-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/jared/git/rtl8188ce-linux-driver/rtl8188ee'
make: *** [all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)
```

Driver built successfully when I switched to the `generic-3.19.x` branch.

FreedomBen, 2015-12-22:
Thanks @jaredharley .  Has the driver (on `generic-3.19.x` branch) been working and stable for you?

jaredharley, 2015-12-22:
@FreedomBen it has, the only problem I've been experiencing is on the 'guest' network at my work - everything works fine until the computer migrates to a different AP node. The connection to 'guest' stays up, but I no longer have internet access. The only resolution is to turn off the wifi, wait a bit, and then turn the wifi back on. This happens with the standard RTL8188EE driver as well, and when I was hunting for a solution online, most people with similar issues resolved it by replacing the wifi AP hardware (obviously not an option for me, at work).