FreedomBen/rtl8188ce-linux-driver #83: [Fedora 22] Building with DKMS

hAh0L13, 2015-08-28:
First, i'm install in automatic mode (git clone + install.sh). But after kernel update i try to use semi-automatic with DKMS. All dependencies has been installed, and i try to run command:
```
$ sudo dkms add rtlwifi/1.0.0 .
[sudo] пароль для hah0l: 
Error! Could not find module source directory.
Directory: /usr/src/rtlwifi-1.0.0 does not exist.
 $ ls /usr/src/ | grep wi
 $ ls /usr/src/ | grep rtl
 $
```
So how i can use dkms for automatic recompile when kernel update?

FreedomBen, 2015-09-14:
Sorry @hAh0L13 , I'm not all that familiar with DKMS.  It's possible that @saiarcot895 may be able to help you?

saiarcot895, 2015-09-15:
For some reason, no one ever noticed this, but the command to run should actually be `sudo dkms add .`. Can you try that instead?

hAh0L13, 2015-09-15:
```
$ sudo dkms add . .
[sudo] пароль для hah0l: 
: I do not know how to handle ..
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>

[hah0l@Exploding ~] $ sudo dkms add -m rtlwifi -v 1.0.0
Error! Could not find module source directory.
```

saiarcot895, 2015-09-15:
Only one dot at the end.

    sudo dkms add .

hAh0L13, 2015-09-15:
```
$ sudo dkms add .
[sudo] пароль для hah0l: 
Error! Invalid number of arguments passed.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
```

FacundoAcevedo, 2016-09-17:
I've done a PR with a solution for this problem #98 
You have to copy the repo to /usr/src/  and then you will can use dkms. See my PR you have to do a play with the name.

greets!