FreedomBen/rtl8188ce-linux-driver #85: Weird problems after installing some updates

dengxinyue0420, 2015-10-14:
Hi,
When I first installed 14.04 LTS and used this driver, everything works perfect. However, after I installed some system updates, things get really weird.
I can connect WIFI, and seems okay for 1 hour, and then it disconnects. Sometimes, it seems connected, but extremely slow. If it disconnects, it tries to reconnect several times, sometimes it succeeds, sometimes not.
I have tried to uninstall and reinstall it, and I have tried restarting computers. If I reinstalled it, it seems fine for the first couple hours and then oscillates again. Any idea on this? What information should I give you for debugging this? 
If I uninstall the driver, and restart my computer, it shows no wifi, and I cannot even enable my wifi. 
I have no idea why it doesn't work...
Thank you so much!

dengxinyue0420, 2015-10-14:
My Ubuntu version is 14.04.3 LTS, and my kernel version is 3.19.0-30-generic.
Thank you so much for providing this driver. I really appreciate if you could help me. It drives me crazy...

FreedomBen, 2015-10-15:
Hi @dengxinyue0420 .  That sounds like a `NetworkManager` issue :disappointed: .  When you uninstall the driver, NetworkManager doesn't see the card and thus assumes you have no wifi connection.

It is possible however that the driver is causing the issues, but highly unlikely since the same version of the driver worked fine before the upgrade.  Sorry I wish I could help more.  You might try items 3 and 4 under [troubleshooting](https://github.com/FreedomBen/rtl8188ce-linux-driver#troubleshooting), but otherwise I'm not sure.

I'll leave the issue open in case someone else wants to chime in.

jcdenton2k, 2015-11-16:
What system updates did you install?  Can you go into your install log and tell us?
Also tell us whether that is Ubuntu or a derivative since Linux Mint is based on that same 14.04 LTS version.

Go into your terminal and type:
uname -a

Then copy and paste the output in your response.
Presuming you're on Ubuntu, you likely had the 'backports' option enabled by default.  That is likely the cause of your issue since backports are unstable/untested code.

You'll need to do some troubleshooting on your end to find out exactly what packages were installed that were backports and then 'downgrade' them somehow.  Explaining how to do that is beyond my skill since I always go in and disable 'backports' and 'romeo/unstable' from my Repository Sources whenever I do a fresh distro install.