FreedomBen/rtl8188ce-linux-driver #87: Hard blocked: yes

zerutreck, 2015-11-28:
Hello, after install without any warning message, wifi didnt work.

Distributor ID:	Ubuntu
Description:	Ubuntu 15.10
Release:	15.10
Codename:	wily
Linux popy-Infoway 4.2.0-18-generic #22-Ubuntu SMP Fri Nov 6 23:32:28 UTC 2015 i686 i686 i686 GNU/Linux
btrtl                  16384  1 btusb
bluetooth             466944  24 bnep,btbcm,btrtl,btusb,btintel
rtl8192se              61440  0
rtl8192ce              53248  0
rtl8192c_common        49152  1 rtl8192ce
rtlwifi                94208  3 rtl8192c_common,rtl8192ce,rtl8192se
mac80211              659456  3 rtlwifi,rtl8192ce,rtl8192se
cfg80211              483328  2 mac80211,rtlwifi
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: yes
1: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no

./am_i_using_this_driver.sh 
[*] You are running the new rtlwifi
[*] You are running the new rtl8192se

FreedomBen, 2015-11-28:
Looks like you have a hardware (or emulated hardware) switch disabling your wifi card.

zerutreck, 2015-11-28:
Yes, but the light of wifi symbol its on. And i havent a button to turn off
or turn on.

FreedomBen, 2015-11-29:
Sometimes it is triggered by holding a modifier key and pressing an F-key, or some other keyboard combination.  It may be a BIOS setting too.  On my wife's computer, it was hiding under the battery cover :facepunch: 

I have heard of rare cases where the system has a BIOS setting or something that Windows enables at boot time and disables at shutdown time.  The solution is to boot Windows and then yank the power.  Good luck!