FreedomBen/rtl8188ce-linux-driver #89: modprobe: ERROR: could not insert 'rtl8192ce': Invalid argument

Fruneng, 2016-01-19:
```
# uname -a
Linux gentoo-cc 4.1.12-gentoo #1 SMP Tue Jan 19 01:36:07 CST 2016 x86_64 Intel(R) Core(TM) i5-2450M CPU @ 2.50GHz GenuineIntel GNU/Linux
# lspci |grep Network
08:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter (rev 01)
# git branch
* generic-4.1.x
  master
# ./am_i_using_this_driver.sh 
[*] You are running the new rtlwifi
[*] You are running the new rtl8192c_common
[*] Not running any rtl8192ce
 # modprobe rtl8192ce
modprobe: ERROR: could not insert 'rtl8192ce': Invalid argument

```

Fruneng, 2016-01-20:
I found the same problem.
http://askubuntu.com/questions/260991/rtl8188ce-wifi-adapter-is-not-detected-after-installing-newest-realtek-driver

closed 2016-01-20 by Fruneng

FreedomBen, 2016-01-20:
Thanks @Fruneng .  Was that file written by the packaged installer?

Fruneng, 2016-01-20:
Yes.I installed the official driver, firstly.  @FreedomBen

ravignelyf, 2019-10-24:
delete modprobe folder from root and then repeat the same code it will work