FreedomBen/rtl8188ce-linux-driver #90: How to improve performance of rtl8192ce driver?

Confusion, 2016-01-23:
Hi, I have RTL8192CE hardware and am experiencing packet loss and fluctuating throughput (with a low average) with either the kernel driver or your updated driver. I have a spare USB WLAN-dongle (a Ralink RT3072) that I used to determine it's the rtl8192ce hardware or the driver.

How can I help to further diagnose the problem? My hardware is a 'true'
> 02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8192CE PCIe Wireless Network Adapter (rev 01)

I've tried enabling debug information, but nothing seems to be logged.

FreedomBen, 2016-01-24:
Hi @Confusion .  You're embarking down a difficult path, but if you do it you'll benefit a lot from the knowledge.  This is a great resource to get you started:  http://www.xml.com/ldd/chapter/book/ch04.html

It's a little old but still pretty relevant.  You might also check out this post: https://www.linux.com/learn/linux-training/33991-the-kernel-newbie-corner-kernel-and-module-debugging-with-gdb

FreedomBen, 2016-01-24:
Specifically tho, you'll see that there are no easy performance improvements.  It's been a while since I profiled, but I remember there being some slowness in encryption as a result of hardware issues.  The encryption can be done in software (and is the recommended setting) but this comes with a performance hit.  You'll also see different results based on the firmware of the router you're connecting to, as some routers conform to the IEEE 802.11 spec better than others.

jcdenton2k, 2016-04-17:
Necro-bump for relevant useful tip to post for anyone pondering this same issue.
Full-speed Wireless-N speeds (or higher) require you to ENABLE 'Quality of Service' (QoS) on your router and to leave it enabled at all times.
Apparently part of the high-speed wireless specs beyond Wireless-G have it baked-in to require QoS to be enabled before it will utilize those speeds.

Since Linux distros are spec-compliant then we get the current situation here.  I do know that QoS doesn't degrade performance if you are the only one using the router, but if multiple people are using the router then it will do what it can to prioritize multimedia traffic (video/streaming) compared to standard data-based traffic (loading a website).  Depending on the router, this can all be configured and customized as needed.

So before digging around with dkms manipulation and modules, might wanna check the router and make sure it has QoS enabled and working properly.  If you don't know what to do or how to configure it then contact your ISP and ask them to help you with it.  That's part of why you pay them monthly.