FreedomBen/rtl8188ce-linux-driver #91: Ubuntu 16.04: Stock Drivers Continue Running

mmstick, 2016-04-15:
Seems that after compiling and installing on Ubuntu 16.04, even though the modules are installed in **/lib/modules/4.4.0-18-generic/kernel/drivers/net/wireless/rtlwifi**, the stock drivers continue to run instead of these drivers.

jcdenton2k, 2016-04-17:
According to the official Ubuntu release info:
http://releases.ubuntu.com/

The latest supported version is 15.10 (Wily Werewolf).
The version you reference (16.04) is listed as unstable/beta and likely won't be supported by most software, let alone these wireless drivers.  Since that version is constantly in flux and changing, it is wise to wait for a 'final' stable release of that version before testing it.

When I checked the link, it linked me to this:
http://releases.ubuntu.com/16.04/

And that says 'Beta 2' which means it isn't considered a stable distro.

According to kernel.org, the latest stable kernel is 4.5.1 (aka Ubuntu Wily) rather than 4.6 and the 4.6 version is still in early RC (release candidate) status.

The PHB Crystal Ball says we are likely to get a stable/final v4.6 kernel near the end of May:
http://phb-crystal-ball.org/

So until then, hang onto v4.5.1 and enjoy it :)
I've successfully tested and installed the v4.5.1 kernel and compiled these drivers on Linux Mint 17.3 (aka Ubuntu  14.04 LTS) and everything works fine.

mmstick, 2016-04-18:
Your information is invalid and incorrect, and you would be ridiculously naiive to assume that I am not already aware that Ubuntu 16.04 hasn't formally released yet despite the fact that it **releases in three days** and thus already went into a **feature freeze** well over a month ago.

Ubuntu has something called a **feature freeze** whereby new versions of software will not be added to the software repository, which thereby means that **Ubuntu 15.10 does not carry Linux 4.5**, and neither does Ubuntu 16.04 which has also reached it's feature freeze. Canonical does not follow kernel.org and never upgrades the Linux kernel point releases after it is shipped. Ubuntu 16.04 is shipping with kernel 4.4 and that is final.

The 04 in Ubuntu 16.04 stands for the month that the distribution will make it's initial release, which is this month in **exactly three days**, where all users of Ubuntu, LTS and otherwise, will be advise to upgrade immediately.

Ubuntu 16.04 is not unstable and I am confused as to why you think that most software isn't supported on Ubuntu 16.04, development branch or otherwise, as Canonical doesn't start from scratch for each new release of Ubuntu but merely pull in more packages from Debian Sid's repository. It therefore has support for more software than any previous release of Ubuntu.

jcdenton2k, 2016-05-08:
I'm gonna spare you the wall of text by keeping it simple.
Feature freeze doesn't mean that the software isn't buggy.  It still needs to be bugtested before it comes out of 'beta' status.  At the time that I wrote my original comment, that release was still in 'beta' status.
It seems that the websites that I linked (Ubuntu's official websites) have finally been updated to remove the beta status.

I commented originally on the 17th of April and it wasn't until the 21st of April that the LTS release was officially announced.  At the time of my original comment, the linked Ubuntu websites were displaying the 'Beta' status that I pointed towards in regards to stability.
http://insights.ubuntu.com/2016/04/20/canonical-unveils-6th-lts-release-of-ubuntu-with-16-04/

That said, if you want FreedomBen to utilize support for 16.04 then by all means feel free to help them implement such support with bugfixes.

I'm concerned when people expect support for RC kernels and such, but a new LTS release should be supported soon once FreedomBen gets time to do it.  I'm fine myself on Linux Mint but for those who want to test it out they are more than welcome to help em out.

ihercowitz, 2016-11-03:
Well, I had the same problem... I'm using Ubuntu 16.04 LTS and I tried to install (and uninstall) the driver several times and the driver was always the Stock one...

So, I ran the command `modinfo rtlwifi` and saw that the driver Ubuntu was using was a dkms one  (rtlwifi-new-dkms). After remove this one (apt-get remove rtlwifi-new-dkms), the rtl8188ce-linux-driver started perfectly! :)

 I hope this could help :)

Best regards!

jcdenton2k, 2016-11-03:
Thank you for the update.  Much appreciated!  :)