FreedomBen/rtl8188ce-linux-driver #101: DKMS Error Ubuntu 16.04

etem, 2016-11-19:
Hello,

I trying to build with DKMS i get this error:
"The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built."

I can confirm that the branch matches the kernel : 
Your branch is up-to-date with 'origin/generic-4.8.x'.

Ubuntu 16.04 - amd64
4.8.0-040800-generic

FreedomBen, 2016-11-22:
Are you using the 4.8 kernel on Ubuntu or the 4.4?  Also can you verify the branch you are on?

I did just find a bug in the ubuntu-16.04 branch with the dkms BUILD EXCLUSIVE.  It was at 4.3 so I bumped it to 4.4 based on https://wiki.ubuntu.com/XenialXerus/ReleaseNotes

If you're using the 4.8 kernel on Ubuntu, the generic-4.8.x branch (that you appear to be using) may work but I'm not sure.  Ubuntu uses some weird hacked up kernels (hence the reason ubuntu routinely gets its own branch).

etem, 2016-11-23:
Hi Ben.

Yes I am using manually installed 4.8 kernel.
In my previous attempts automatic installation didn't worked with stock kernel - 4.4
I will revert back to the stock kernel and try DKMS method again and will post the results.

etem, 2016-11-25:
Hi,

DKMS installation fails also with stock kernel - 4.4.0-47-generic : 

bavarian@ThinkPad-T530:~/rtl8188ce-linux-driver$ sudo dkms install rtlwifi/1.0.0Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.

elKaZe, 2016-11-27:
Same problem in fedora, I've to replace with 4.8.6, to avoid it.

etem, 2016-12-04:
@elKaZe  Does it works with kernel 4.8.6?

elKaZe, 2017-01-11:
sorry. no, it's not working

awolfmann, 2017-05-26:
Is there any update on this?
I tried to install it by any of the method (automatic, semi automatic and manually)  and i can't get it working.. i'm still using the stock driver and i'm losing my connection every 3-5 minutes.

I have rtl8188ce, ubuntu 16.04 with kernel 4.8.0-36.

Thanks in advance!

VitesWen, 2017-09-04:
I have encountered this issue on Ubuntu 16.10, that still report an error not match with this kernel.do you have resolved it?
------------------------------
Deleting module version: 1.4.20.3+svn502
completely from the DKMS tree.
------------------------------
Done.
Loading new iscsitarget-1.4.20.3+svn502 DKMS files...
First Installation: checking all kernels...
Building only for 4.8.0-22-generic
Building initial module for 4.8.0-22-generic
Error!  The dkms.conf for this module includes a BUILD_EXCLUSIVE directive which
does not match this kernel/arch.  This indicates that it should not be built.
Skipped.
Processing triggers for ureadahead (0.100.0-19) ...
Setting up iscsitarget (1.4.20.3+svn502-2ubuntu4.4) ...
Processing triggers for systemd (231-9git1) ...
Processing triggers for man-db (2.7.5-1) ...
Processing triggers for ureadahead (0.100.0-19) ...

FreedomBen, 2017-09-04:
@VitesWen I'm not sure why this happens, but if you're certain you're using kernel 4.8, can you try removing this line from `dkms.conf` ?

    BUILD_EXCLUSIVE_KERNEL=^4.8.x

FFY00, 2017-09-04:
@FreedomBen the problem is that after editing/removing that line, we get

`Error!  Build of rtl8723ae.ko failed for: 4.x.x`

VitesWen, 2017-09-05:
Hi Ben
Thanks for your comments, I'm not sure the version 1.4.20.3+svn502-2ubuntu4.4
 whether support kernel 4.8.0. it still failed after
removed BUILD_EXCLUSIVE_KERNEL=^4.8.x.

root@test:/etc/apt# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.10
Release:        16.10
Codename:       yakkety
root@test:/etc/apt# uname -a
Linux test 4.8.0-22-generic #24-Ubuntu SMP Sat Oct 8 09:15:00 UTC 2016
x86_64 x86_64 x86_64 GNU/Linux

-- 

Thanks & Regards
Wen Hua
Global Storage Design Vertical
Celestica Technology Consultancy (Shanghai) Co., Ltd  加弘科技咨询(上海)有限公司
3000 Longdong Ave.Building 1, Section A 701-801 Zhangjiang River Front
Harbor Pudong, Shanghai,P.R.C. 201203
Tel: +86-21-61006028 EXT 7374 Email: hwen@celestica.com
<bduan@celestica.com>
www.celestica.com

Solid Partners, Flexible Solutions

iafilatov, 2017-11-11:
`BUILD_EXCLUSIVE_KERNEL` should be a regex, that's why values like `^4.8.x` never match anything. Changing it to `^4\.8\.` help. And I think this should be done in all branched in this repo with `x` in the value (some have `*` which is ok but a more precise regex would be better still).
```
       BUILD_EXCLUSIVE_KERNEL=
              This   optional  directive  allows  you  to  specify  a  regular
              expression which defines the subset of  kernels  which  DKMS  is
              allowed to build your module for.  If the kernel being built for
              does not match against this regular expression, the  dkms  build
              will  error  out.  For example, if you set it as ="^2.4.*", your
              module would not be built for 2.6 kernels.
```

Second problem is `verify_branch.sh` and `backup_existing.sh` which can ask for user interaction during dkms build. Remove/rename them in `/usr/src/rtlwifi-1.0.0`.

Third problem is that in some branches `dkms.conf` contains more modules than `Makefile`. DKMS runs `make`, checks if all expected drivers have been built and fails. Either `Makefile` should contain all modules of `dkms.conf` should contain only the ones from `Makefile`.