FreedomBen/rtl8188ce-linux-driver #103: exec format error

julianofischer, 2016-12-04:
Hello,
I tried automatic installation but unsuccessfully (#75);

So, I tried to copy the compiled drivers manually:

    cp rtlwifi.ko /lib/modules/4.4.0-51-generic/kernel/drivers/net/wireless/realtek/rtlwifi/
    cp rtl8192c/rtl8192c_common.ko /lib/modules/4.4.0-51-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192c/rtl8192c-common.ko
    cp rtl8192ce/rtl8192ce.ko /lib/modules/4.4.0-51-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192ce/rtl8192ce.ko

   ./am_i_using_this_driver.sh 
   [*] You are running the new rtlwifi
   [*] You are running the new rtl8192c_common
   [*] Not running any rtl8192ce

   sudo modprobe rtl8192ce 
   modprobe: ERROR: could not insert 'rtl8192ce': Exec format error

   sudo modprobe -f rtl8192ce
   modprobe: ERROR: could not insert 'rtl8192ce': Exec format error

I think realtek binaries (firmware) are the problem.

julianofischer, 2016-12-04:
After reinstalling kernel
    sudo apt-get install --reinstall linux-image-generic linux-image-4.4.0-51-generic 

The output is:

    ./am_i_using_this_driver.sh 
    [*] Not running the new rtlwifi (running stock driver)
    [*] Not running the new rtl8192c_common (running stock driver)
    [*] You are running the new rtl8192ce

if I try to copy the other modules

    sudo cp rtlwifi.ko /lib/modules/4.4.0-51-generic/kernel/drivers/net/wireless/realtek/rtlwifi/
    sudo cp rtl8192c/rtl8192c_common.ko /lib/modules/4.4.0-51-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtl8192c/rtl8192c-common.ko

I return to the same problem

    ./reload_modules.sh 
    modprobe: ERROR: could not insert 'rtl8192ce': Exec format error
    [*] You are running the new rtlwifi
    [*] You are running the new rtl8192c_common
    [*] Not running any rtl8192ce

vaclavhodek, 2017-06-08:
I had the same problem with _Exec format error_ and invoking **make clean** before **./install.sh** solved it.