FreedomBen/rtl8188ce-linux-driver #104: New modules not being used (perhaps due to different folder structure)

beersteiner, 2016-12-16:
Ben,
I'm having an issue trying to use this for the rtl8188ee driver.  I followed your instructions using both the ./install.sh method as well as the manual method.  Everything works fine but I am when I check if the new driver is being used, it is not.  If i try "sudo modprobe rtl8188ee" it appears to execute, but the new driver still isn't used.  I suspect I know why... I believe my system places these modules in "/lib/modules/4.4.0-53-generic/kernel/drivers/net/wireless/realtek/rtlwifi/" whereas I suspect your code assumes it is in "/lib/modules/4.4.0-53-generic/kernel/drivers/net/wireless/rtlwifi/".  Therefore, I suspect that when I modprobe in rtl8188ee, it successfully loads the one already being used in the realtek/rtlwifi folder.

I'm relatively new to linux, and I don't want to mess anything up.  How do you suggest I fix this?  Should I change your install file somewhere to set the directory structure correctly?  Or is there an easier way to accommodate this difference?  Or do I have it all wrong?  Thanks very much!

FreedomBen, 2016-12-17:
Hi John!  Welcome to Linux!

BUILD_EXCLUSIVE_KERNEL does sometimes get set incorrectly.  If you're using generic-4.4.x branch on a 4.4 kernel, you can definitely change the BUILD_EXCLUSIVE_KERNEL to 4.3.

As far as destination of the built modules, I've never seen them in a different place but that is entirely possible.  I would backup (copy) the old ones just in case tho.  The build script usually does this for you, but if the modules are not where it expects then it will not be backing them up properly.

What distro are you using?

FreedomBen, 2016-12-17:
I did just check the generic 4.4 branch and the dkms.conf BUILD_EXCLUSIVE_KERNEL is 4.4.x.  If it's not on yours, perhaps pull latest? 

    git pull --rebase

beersteiner, 2016-12-17:
I wasn't asking to change the BUILD_EXCLUSIVE_KERNEL to 4.3.x, I was asking to change it to 4.4.x - I was apparently looking at the wrong branch of your files.  I'm not on the machine right now, but when I get back on it I'll confirm this is set to 4.4.x after checking the correct branch out.  If that's the case, then I don't know why I'm getting the error.  I'm on definitely on Ubuntu 16.04, kernel 4.4.0-53-generic.

FreedomBen, 2016-12-17:
D'oh I'm sorry for the confusion.  I typoed in a bad spot (note the number on the end).  I edited the original post from:

> BUILD_EXCLUSIVE_KERNEL does sometimes get set incorrectly. If you're using generic-4.4.x branch on a 4.4 kernel, you can definitely change the BUILD_EXCLUSIVE_KERNEL to **4.3**.

to:

> BUILD_EXCLUSIVE_KERNEL does sometimes get set incorrectly. If you're using generic-4.4.x branch on a 4.4 kernel, you can definitely change the BUILD_EXCLUSIVE_KERNEL to **4.4**.