FreedomBen/rtl8188ce-linux-driver #105: the new modules seems not being used

fivestrong, 2016-12-27:
Hi , my wireless drops frequently recently.May be it was the problem of Realtek.
this the type of realtek.
```
lspci -k | grep Network
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8188CE 802.11b/g/n WiFi Adapter (rev 01)
```
my system is archlinux  
```
uname -r 
4.4.39-1-MANJARO
```
then i follow the Manual Installation step by step,but when all was finished. i run the command
`./am_i_using_this_driver.sh`
it show like this:
```
[*] Not running the new rtlwifi (running stock driver)
[*] Not running the new rtl8192c_common (running stock driver)
[*] Not running the new rtl8192ce (running stock driver)
```
and seem like  the system still use the other mod
```
modinfo rtlwifi 
filename:       /lib/modules/4.4.39-1-MANJARO/kernel/drivers/net/wireless/realtek/rtlwifi/rtlwifi.ko.gz
description:    Realtek 802.11n PCI wireless core
license:        GPL
author:         Larry Finger    <Larry.FInger@lwfinger.net>
author:         Realtek WlanFAE <wlanfae@realtek.com>
author:         lizhaoming      <chaoming_li@realsil.com.cn>
depends:        mac80211,cfg80211
intree:         Y
vermagic:       4.4.39-1-MANJARO SMP preempt mod_unload modversions 
```
the files we compiled list here
```
ls /lib/modules/4.4.39-1-MANJARO/kernel/drivers/net/wireless/rtlwifi 
rtl8188ee  rtl8192c  rtl8192ce  rtl8192cu  rtl8192de  rtl8192se  rtlwifi.ko
```
it may should be here

```
ls /lib/modules/4.4.39-1-MANJARO/kernel/drivers/net/wireless/realtek/rtlwifi 
btcoexist  rtl8192c   rtl8192cu  rtl8192ee  rtl8723ae  rtl8723com  rtl_pci.ko.gz  
rtl8188ee  rtl8192ce  rtl8192de  rtl8192se  rtl8723be  rtl8821ae   rtl_usb.ko.gz  rtlwifi.ko.gz
```
i have tried to remove them all and repleace with our files,but it didn't work,the mod cannot set up .

Anyone have ideas how can i fix this problem?

fivestrong, 2016-12-28:
I have solve the problem:
just remove the  `/lib/modules/4.4.39-1-MANJARO/kernel/drivers/net/wireless/realtek/rtlwifi/*` and replace with the  cotent in
`/lib/modules/4.4.39-1-MANJARO/kernel/drivers/net/wireless/rtlwifi/*`
and the most importent you need run this commond:
`sudo depmod -a
`
this will rebuild the dependence of the mod we build and then 
`sudo modprobe rtl8192ce`

for me it works well:
```
rtl8188ce-linux-driver git:(generic-4.4.x) ./am_i_using_this_driver.sh                                     
[*] You are running the new rtlwifi
[*] You are running the new rtl8192c_common
[*] You are running the new rtl8192ce

```

FreedomBen, 2017-01-08:
awesome, thanks for following up!

closed 2017-01-08 by FreedomBen