FreedomBen/rtl8188ce-linux-driver #111: Can't install via DKMS on Fedora 25; DKMS thinks build failed when it didn't

fat-lobyte, 2017-01-30:
Hi! I have been trying to install the driver on Fedora via DKMS. I am running  the Fedora Kernel with the version `4.9.5-200.fc25.x86_64`.

I got the source, switched to the `generic-4.9.x` branch and followed the Instructions in the README.md.
I received an error described in issue #110, but I fixed it by editing the BUILD_EXCLUSIVE directive in `dkms.conf`.

However, according to DKMS the build "failed":

# dkms install rtlwifi/1.0.0

Kernel preparation unnecessary for this kernel.  Skipping...

```
Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.9.5-200.fc25.x86_64 KVER=4.9.5-200.fc25.x86_64................
Error!  Build of rtl8723ae.ko failed for: 4.9.5-200.fc25.x86_64 (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/rtlwifi/1.0.0/build/ for more information.
```

The make.log file is attached here: 
[make.log.txt](https://github.com/FreedomBen/rtl8188ce-linux-driver/files/739123/make.log.txt)

The strange thing is: the makefile does *not* contain any errors! All modules seem to have been successfully built, but DKMS somehow decided it was a failure.

Any Idea what's wrong with that?

jvmCarneiro, 2017-05-06:
I can confirm the exact same situation (including the dkms.conf fix) on Arch, both in latest `4.10.13-1-ARCH` and lts `4.9.26-1-ARCH` kernels.

FreedomBen, 2017-05-12:
Thanks for the reports.  I have no idea what's wrong here either, but I am a DKMS newb.  It's possible that @saiarcot895 might have some ideas.