FreedomBen/rtl8188ce-linux-driver #113: Build on Centos failed

fadnincx, 2017-02-04:
Build on Centos 7 failed with the following error:

make -C /lib/modules/3.10.0-514.6.1.el7.x86_64/build M=/root/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/kernels/3.10.0-514.6.1.el7.x86_64'
  CC [M]  /root/rtl8188ce-linux-driver/base.o
In file included from /root/rtl8188ce-linux-driver/base.c:33:0:
/root/rtl8188ce-linux-driver/wifi.h:1369:40: error: ‘IEEE80211_NUM_BANDS' undeclared here (not in a function)
  struct ieee80211_supported_band bands[IEEE80211_NUM_BANDS];
                                        ^
/root/rtl8188ce-linux-driver/base.c:141:10: error: 'IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
  .band = IEEE80211_BAND_2GHZ,
          ^
/root/rtl8188ce-linux-driver/base.c:153:10: error: 'IEEE80211_BAND_5GHZ' undeclared here (not in a function)
  .band = IEEE80211_BAND_5GHZ,
          ^
make[2]: *** [/root/rtl8188ce-linux-driver/base.o] Error 1
make[1]: *** [_module_/root/rtl8188ce-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/3.10.0-514.6.1.el7.x86_64'
make: *** [all] Error 2

I get no error on compile the master branch, but theres a problem with setting the driver.

There should be all dependencies installed.

FreedomBen, 2017-02-08:
@fadnincx Is this a fully updated system?  CentOS 7.3?

drkhosla, 2017-08-05:
I am facing same error. I am also having 7.3.1611. Any solution

albpal, 2017-10-14:
I'm not able either.

**Kernel**:
3.10.0-693.2.2.el7.x86_64
**SO**
CentOS Linux release 7.4.1708

**Error logs**:
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:84:12: error: ‘IEEE80211_BAND_2GHZ’ undeclared here (not in a function)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:93:12: error: ‘IEEE80211_BAND_5GHZ’ undeclared here (not in a function)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:159:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:159:2: error: (near initialization for ‘rtw_5ghz_a_channels[0].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:159:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:159:2: error: (near initialization for ‘rtw_5ghz_a_channels[1].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:160:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:160:2: error: (near initialization for ‘rtw_5ghz_a_channels[2].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:160:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:160:2: error: (near initialization for ‘rtw_5ghz_a_channels[3].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:161:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:161:2: error: (near initialization for ‘rtw_5ghz_a_channels[4].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:161:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:161:2: error: (near initialization for ‘rtw_5ghz_a_channels[5].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:162:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:162:2: error: (near initialization for ‘rtw_5ghz_a_channels[6].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:162:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:162:2: error: (near initialization for ‘rtw_5ghz_a_channels[7].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:163:2: error: initializer element is not constant
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:163:2: error: (near initialization for ‘rtw_5ghz_a_channels[8].band’)
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:163:2: error: initializer [... cut ...]
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:6461:38: error: array subscript is not an integer
/root/EW-7611ULB_Wireless_Driver/RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844/driver/rtl8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX20150119-5844/os_dep/linux/ioctl_cfg80211.c:6462:38: error: array subscript is not an integer
Compile make driver error: 2
Please check error Mesg