FreedomBen/rtl8188ce-linux-driver #121: Installs to wrong directory (and doesn't work)

x0a, 2017-07-13:
OS: Ubuntu 16.04.2
Kernel: 4.4.0-83-generic
Wireless card: 

>            *-network UNCLAIMED
>                 description: Network controller
>                 product: RTL8188CE 802.11b/g/n WiFi Adapter
>                 vendor: Realtek Semiconductor Co., Ltd.
>                 physical id: 0
>                 bus info: pci@0000:01:00.0
>                 version: 01
>                 width: 64 bits
>                 clock: 33MHz
>                 capabilities: pm msi pciexpress bus_master cap_list
>                 configuration: latency=0
>                 resources: ioport:3000(size=256) memory:f0200000-f0203fff

I've been using this driver for about 2 years with no issues, cloning & recompiling after every update. But I stopped using it for about 5 months because I got a wired connection. I tried to install it recently but it failed every time (./am_i_using_this_driver.sh showed I was still running the stock driver, persisted after reboot).

In Makefile, the destination directory is defined as:
> MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/rtlwifi

But the drivers are actually in
/lib/modules/$(KVER)/kernel/drivers/net/wireless/**realtek**/rtlwifi

If I update the directory, the drivers successfully install and are detected by ./am_i_using_this_driver.sh. However, once I turn wifi on, everything freezes, keyboard becomes unresponsive and I'm forced to hard reset my laptop. On boot, I see a bunch of "[FAILED] Failed to start Load Kernel Modules" messages and wifi doesn't show up in the notification bar. Let me know how I can help debug this.

x0a, 2017-07-18:
OK I was finally able to get it to install and run by more thoroughly replacing all instances of **wireless/rtlwifi** with **wireless/realtek/rtlwifi**

The quick and dirty way to get it to install is to change line 41 in **./install.sh** from
`make && make install`
to
`make && grep --exclude={"dkms.conf","base.c.log","rebase.sh","index","README.md","build-and-modprobe.rb"} -rl "wireless/rtlwifi" | xargs sed -i "s/wireless\/rtlwifi/wireless\/realtek\/rtlwifi/g" && make install`

Of course the more permanent solution is to parse output from "modprobe rtl*" to get the real location of the module instead of hardcoding.

Aside from that, thank you for the excellent work on this driver!

FFY00, 2017-09-04:
@x0a after applying this fix the computer freezes like you described and my keyboard's Caps Lock and Scroll Block start flashing.

zboyd2, 2017-11-19:
I ran into a very similar issue and eventually fixed it by deleting everything in the realtek directory and replacing it with a symbolic link to the directory where the install script put everything. I also had to put in a script in init.d to load the necessary modules at each boot, since modprobe does not like this solution for some reason..

LocoUnPocoNadaMas, 2020-11-18:
> OK I was finally able to get it to install and run by more thoroughly replacing all instances of **wireless/rtlwifi** with **wireless/realtek/rtlwifi**
> 
> The quick and dirty way to get it to install is to change line 41 in **./install.sh** from
> `make && make install`
> to
> `make && grep --exclude={"dkms.conf","base.c.log","rebase.sh","index","README.md","build-and-modprobe.rb"} -rl "wireless/rtlwifi" | xargs sed -i "s/wireless\/rtlwifi/wireless\/realtek\/rtlwifi/g" && make install`
> 
> Of course the more permanent solution is to parse output from "modinfo rtl*" to get the real location of the module instead of hardcoding.
> 
> Aside from that, thank you for the excellent work on this driver!

thanks mate!