FreedomBen/rtl8188ce-linux-driver #123: build fails with "implicit declaration of function 'RT_ASSERT'" in rtl8188ee/pwrseqcml.c:143

briansorahan, 2017-07-26:
I'm building on 4.11.9-1-ARCH

tony, 2017-08-18:
```
~/Downloads/rtl8188ce-linux-driver generic-4.12.x 13s
❯ sudo ./install.sh           

[*] So you want to live on the wild side and try a different driver for your RealTek wireless card eh?  Awesome!  I'll help you do it.

[*] We are going to build and install the driver from source code, compiled specifically for your machine.

[*] If you want to do the build/install manually, there are instructions in the "README.md" file.

[*] Please report any bugs/problems at https://github.com/FreedomBen/rtl8188ce-linux-driver

[*] Press <Enter> when ready to begin, or <Ctrl+C> to quit

[*] Let's install any dependencies you will need in order to build the driver.
warning: git-2.14.1-1 is up to date -- skipping
 there is nothing to do
warning: linux-headers-4.12.8-1 is up to date -- skipping
 there is nothing to do
:: There are 25 members in group base-devel:
:: Repository core
   1) autoconf  2) automake  3) binutils  4) bison  5) fakeroot  6) file
   7) findutils  8) flex  9) gawk  10) gcc  11) gettext  12) grep  13) groff
   14) gzip  15) libtool  16) m4  17) make  18) pacman  19) patch
   20) pkg-config  21) sed  22) sudo  23) texinfo  24) util-linux  25) which

Enter a selection (default=all): 
warning: autoconf-2.69-4 is up to date -- skipping
warning: automake-1.15.1-1 is up to date -- skipping
warning: binutils-2.28.0-4 is up to date -- skipping
warning: bison-3.0.4-3 is up to date -- skipping
warning: fakeroot-1.21-2 is up to date -- skipping
warning: file-5.31-1 is up to date -- skipping
warning: findutils-4.6.0-2 is up to date -- skipping
warning: flex-2.6.4-1 is up to date -- skipping
warning: gawk-4.1.4-2 is up to date -- skipping
warning: gcc-7.1.1-4 is up to date -- skipping
warning: gettext-0.19.8.1-2 is up to date -- skipping
warning: grep-3.1-1 is up to date -- skipping
warning: groff-1.22.3-7 is up to date -- skipping
warning: gzip-1.8-2 is up to date -- skipping
warning: libtool-2.4.6-8 is up to date -- skipping
warning: m4-1.4.18-1 is up to date -- skipping
warning: make-4.2.1-2 is up to date -- skipping
warning: pacman-5.0.2-2 is up to date -- skipping
warning: patch-2.7.5-1 is up to date -- skipping
warning: pkg-config-0.29.2-1 is up to date -- skipping
warning: sed-4.4-1 is up to date -- skipping
warning: sudo-1.8.20.p2-1 is up to date -- skipping
warning: texinfo-6.4-1 is up to date -- skipping
warning: util-linux-2.30.1-2 is up to date -- skipping
warning: which-2.21-2 is up to date -- skipping
 there is nothing to do

[*] Now let's compile the driver from source and copy the files to the right directories
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
Yes
make -C /lib/modules/4.12.8-1-ARCH/build M=/home/x/Downloads/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/lib/modules/4.12.8-1-ARCH/build'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/lib/modules/4.12.8-1-ARCH/build'
make[1]: Entering directory '/home/x/Downloads/rtl8188ce-linux-driver/rtl8192c'
make -C /lib/modules/4.12.8-1-ARCH/build M=/home/x/Downloads/rtl8188ce-linux-driver/rtl8192c modules
make[2]: Entering directory '/usr/lib/modules/4.12.8-1-ARCH/build'
  Building modules, stage 2.
[... 29 lines cut ...]
make[2]: Leaving directory '/usr/lib/modules/4.12.8-1-ARCH/build'
make[1]: Leaving directory '/home/x/Downloads/rtl8188ce-linux-driver/rtl8192de'
make[1]: Entering directory '/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee'
make -C /lib/modules/4.12.8-1-ARCH/build M=/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee modules
make[2]: Entering directory '/usr/lib/modules/4.12.8-1-ARCH/build'
  CC [M]  /home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o
/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c: In function ‘rtl88_hal_pwrseqcmdparsing’:
/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:143:5: error: implicit declaration of function ‘RT_ASSERT’; did you mean ‘RTA_NEXT’? [-Werror=implicit-function-declaration]
     RT_ASSERT( false,
     ^~~~~~~~~
     RTA_NEXT
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:303: /home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Error 1
make[2]: *** [Makefile:1515: _module_/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.12.8-1-ARCH/build'
make[1]: *** [Makefile:33: default] Error 2
make[1]: Leaving directory '/home/x/Downloads/rtl8188ce-linux-driver/rtl8188ee'
make: *** [Makefile:41: all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)

```

4.12.8-1-ARCH

(antergos)

AdamHutchison, 2017-08-28:
I solved this in ubuntu by making sure that my kernel version matched the release version stated in the tables shown under the Manual installation instructions. Hope that helps.

zml, 2017-09-09:
I am having the exact same error on Fedora 26 -**4.12.9-300.fc26.x86_64**. How can I solve it?

jcdenton2k, 2017-09-18:
I've also received a similar error.

`  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/*myusername*/rtl8188ce-linux-driver/rtl8192de/rtl8192de.mod.o
  LD [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8192de/rtl8192de.ko
make[2]: Leaving directory '/usr/src/linux-headers-4.13.2-041302-generic'
make[1]: Leaving directory '/home/*myusername*/rtl8188ce-linux-driver/rtl8192de'
make[1]: Entering directory '/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee'
make -C /lib/modules/4.13.2-041302-generic/build M=/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee modules
make[2]: Entering directory '/usr/src/linux-headers-4.13.2-041302-generic'
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/hw.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/table.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/sw.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/trx.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/led.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/fw.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/phy.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/rf.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/dm.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseq.o
  CC [M]  /home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o
/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c: In function ‘rtl88_hal_pwrseqcmdparsing’:
/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:143:5: error: implicit declaration of function ‘RT_ASSERT’ [-Werror=implicit-function-declaration]
     RT_ASSERT( false,
     ^
cc1: some warnings being treated as errors
scripts/Makefile.build:302: recipe for target '/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o' failed
make[3]: *** [/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Error 1
Makefile:1512: recipe for target '_module_/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee' failed
make[2]: *** [_module_/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.13.2-041302-generic'
Makefile:33: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/*myusername*/rtl8188ce-linux-driver/rtl8188ee'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)
`

jcdenton2k, 2017-09-20:
OS: Linux Mint 18.2 based on Ubuntu 16.04 LTS
Kernel: 4.13.x (above) and currently using the Linux Mint recommended 4.10.x kernel
Just a quick update so @FreedomBen sees this also.
I was on kernel 4.13.x and that was the cause of my issues getting this driver compiled and working (and the above error) due to my OS and kernel combination not playing nicely or missing some dependencies somehow.
I switched back to the recommended kernel and everything works fine.
The recommended branch was Ubuntu 17.X and that was weird but it still works fine.

briansorahan, 2017-09-22:
@FreedomBen Feel free to close this if @jcdenton2k 's suggestion works. I ended up installing Fedora and it just worked without me having to fiddle with the driver.