FreedomBen/rtl8188ce-linux-driver #127: Ubuntu 17.10 rtl8188ee

fk1412, 2017-09-24:
I try to download the drivers but i get an error during the installation. Please some help

FreedomBen, 2017-09-25:
I haven't merged with 17.10's kernel yet, but I'll get to it soon.  Hopefully within the next day or so.

fk1412, 2017-09-25:
Thank's

FreedomBen, 2017-09-26:
@fk1412 I pushed up the 17.10 branch, so it should be working now.  Let me know if you still run into problems.

fk1412, 2017-09-26:
modprobe: FATAL: Module rtlwifi is in use.
It says that at the end, what i have to do ?

fk1412, 2017-09-26:
And when i do ./am_i_using_the_driver,
it tells me that i'm still using the stock driver

fk1412, 2017-09-26:
I solve the problem but my wifi is still very slow after the installation of the new driver, i don't understand

FreedomBen, 2017-09-26:
If you keep getting the Module in use error, you may just want to reboot.  You can try to make sure all modules rtlwifi depends on are unloaded first, then unload rtlwifi.  However it is probably just easier to reboot.

fk1412, 2017-09-26:
yeah i solve this error but my wifi is still very slow

FreedomBen, 2017-09-26:
If you haven't yet, make sure to try steps 3 and 4 under "troubleshooting":

https://github.com/FreedomBen/rtl8188ce-linux-driver#troubleshooting

fk1412, 2017-09-27:
I try but it still doesn't work