FreedomBen/rtl8188ce-linux-driver #132: Not running the new rtlwifi (running stock driver)

denghaoyang666, 2018-09-08:
Deepin15.7，型号8188ce，安装完成了，但是一直不运行，github上提供的方法都试过了，重启、modprobe rtl8192ce、，都没用。

denghaoyang666, 2018-09-09:
解决了，已经在运行了，，参考了https://github.com/FreedomBen/rtl8188ce-linux-driver/issues/121，，，在install.sh中把`make && sudo make install`修改为了`make && grep --exclude={"dkms.conf","base.c.log","rebase.sh","index","README.md","build-and-modprobe.rb"} -rl "wireless/realtek/rtlwifi" | xargs sed -i "s/wireless\/rtlwifi/wireless\/realtek\/rtlwifi/g" && sudo make install`重新安装后，显示已运行新驱动。

但是，网速依旧被限制在200kB/s

Debufgc, 2019-03-23:
请问下 按此命令输入最后出现sed: 无法打开文件 ireless/realtek/rtlwifi/g: 没有那个文件或目录
这个如何怎么解决？ ubuntu-kaili linux 出现一样的错误 Not running the new rtlwifi (running stock driver)