FreedomBen/rtl8188ce-linux-driver #134: install failed in ubuntu18.04

nevsayno, 2019-01-05:
use branch `ubuntu-18.04`,
there are no errors reported at install stage, but when test new module, it appears:

> 
> [*] Not running the new rtlwifi (running stock driver)
> [*] Not running the new rtl8192c_common (running stock driver)
> [*] Not running the new rtl8192ce (running stock driver)

after dig into the source code,  bug fixed with the following change:

```
diff --git a/Makefile b/Makefile
index 54ae3a9..4b909c2 100644
--- a/Makefile
+++ b/Makefile
@@ -2,7 +2,7 @@
 CC = gcc
 KVER  := $(shell uname -r)
 KSRC := /lib/modules/$(KVER)/build
-MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/rtlwifi
+MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/realtek/rtlwifi
 FIRMWAREDIR := /lib/firmware/
 PWD := $(shell pwd)
 CLR_MODULE_FILES := *.mod.c *.mod *.o .*.cmd *.ko *~ .tmp_versions* modules.order Module.symvers
diff --git a/functions.sh b/functions.sh
index 718fe9e..88f260b 100644
--- a/functions.sh
+++ b/functions.sh
@@ -352,7 +352,7 @@ runningStockRtl8821ae ()
     runningAnyRtl8821ae && ! runningOurRtl8821ae
 }
 
-readonly rtlwifi_orig="/lib/modules/$(uname -r)/kernel/drivers/net/wireless/rtlwifi"
+readonly rtlwifi_orig="/lib/modules/$(uname -r)/kernel/drivers/net/wireless/realtek/rtlwifi"
 readonly rtlwifi_backup_dir="$HOME/.rtlwifi-backup"
 readonly rtlwifi_backup_outfile="$rtlwifi_backup_dir/$(uname -r).tar.gz"
 
@@ -387,7 +387,7 @@ backupCurrent ()
             mkdir -p "$rtlwifi_backup_dir"
             tar -czf "$rtlwifi_backup_outfile" -C "$rtlwifi_orig" . > /dev/null 2>&1
         else
-            echo "Could not backup rtlwifi because it could not be found!  Expected at $rtlwifi_orig" >&2
+            echo "Could not backup rtlwifi because it could not be found!  Expected at $rtlwifi_orig"
         fi
     fi
 }
```

kkursor, 2019-09-16:
Confirm this on latest revision. Had to patch makefiles by hand.

FreedomBen, 2019-10-01:
Hey guys, I couldn't test the patch but I went ahead and pushed it for the ubuntu-18.04 branch.  Can somebody pull latest and test it?

commit 9b7d0d15 (2019-10-01): Fix build issue as a result of unique location

Fixes #134