FreedomBen/rtl8188ce-linux-driver #140: Still 'running stock driver' on ubuntu 18.04 using RTL8192CE

CluEleSsUK, 2019-05-28:
Have search previous issues and tried their solutions but to no avail.

`$ modinfo rtlwifi` gives me the following output:

```
filename:       /lib/modules/4.15.0-50-generic/kernel/drivers/net/wireless/realtek/rtlwifi/rtlwifi.ko
description:    Realtek 802.11n PCI wireless core
license:        GPL
author:         Larry Finger	<Larry.FInger@lwfinger.net>
author:         Realtek WlanFAE	<wlanfae@realtek.com>
author:         lizhaoming	<chaoming_li@realsil.com.cn>
srcversion:     4690A14A94EEFAC6540FD6B
depends:        mac80211,cfg80211
retpoline:      Y
intree:         Y
name:           rtlwifi
vermagic:       4.15.0-50-generic SMP mod_unload 
signat:         PKCS#7
signer:         
sig_key:        
sig_hashalgo:   md4
```

which I'm pretty sure is indeed the built-in driver.

My network card output from lshw is:
```
  description: Wireless interface
                product: RTL8192CE PCIe Wireless Network Adapter
                vendor: Realtek Semiconductor Co., Ltd.
                physical id: 0
                bus info: pci@0000:03:00.0
                logical name: wlp3s0
                version: 01
                serial: something
                width: 64 bits
                clock: 33MHz
                capabilities: bus_master cap_list ethernet physical wireless
                configuration: broadcast=yes driver=rtl8192ce driverversion=4.15.0-50-generic firmware=N/A ip=192.168.0.14 latency=0 link=yes multicast=yes wireless=IEEE 802.11
                resources: irq:19 ioport:3000(size=256) memory:a3200000-a3203fff
```

Any ideas?!

CluEleSsUK, 2019-05-28:
I found a solution!
So similar to https://github.com/FreedomBen/rtl8188ce-linux-driver/issues/121, the drivers were being installed in the wrong directory.
To fix things, I did the following
* edited (./rtl8192ce/Makefile)[./rtl8192ce/Makefile] and added /realtek to the end of the var `MODDESTDIR`. (note: you may have to replace rtl8192ce with your own adapter version)
* `cd /lib/modules/` uname -r`/kernel/drivers/net/wireless/realtek` and nuke everything in there
* run the `./install.sh` again

CluEleSsUK, 2019-05-28:
See PR here https://github.com/FreedomBen/rtl8188ce-linux-driver/pull/141

Ayush-Kumar-web, 2023-04-20:
> I found a solution! So similar to #121, the drivers were being installed in the wrong directory. To fix things, I did the following
> 
>     * edit [./rtl8192ce/Makefile](./rtl8192ce/Makefile) and added /realtek to the end of the var `MODDESTDIR`. (note: you may have to replace rtl8192ce with your own adapter version)
> 
>     * `` cd /lib/modules/` uname -r`/kernel/drivers/net/wireless/realtek `` and nuke everything in there
> 
>     * run the `./install.sh` again

after I run ./install.sh again
it gives error that 
Could not backup rtlwifi because it could not be found! Expected at /lib/modules/5.15.0-70-generic/kernel/drivers/net/wireless/rtlwifi