FreedomBen/rtl8188ce-linux-driver #142: ID 0bda:a192 Realtek

ieow, 2019-12-06:
Hi, 
I am trying to use Mecury MW300UH with realtek id below.
Bus 001 Device 002: ID 0bda:a192 Realtek Semiconductor Corp. 

I tried install.sh but still failed to detect the realtek card.
Could I add the ID to the list and re-install?
If I could, which file should I edit? 
Thanks.

FreedomBen, 2019-12-06:
Hi @ieow .  Did you see this message?

```
[*] I wasn't able to find a Realtek card on your machine.  Do you want to proceed anyway? (Y/N)
```

ieow, 2019-12-07:
Yes, I did

FreedomBen, 2019-12-07:
You probably need to add support then to `functions.sh` .  If you want to skip the detection step you could apply this patch.  Copy this to a file and run `git apply <file>`

```
diff --git a/install.sh b/install.sh
index 21a9f09..2178b34 100755
--- a/install.sh
+++ b/install.sh
@@ -7,7 +7,7 @@ else
     . "$(readlink -f functions.sh)"
 fi
 
-if ! $(pciDetectsRealtekCard || usbDetectsRealtekCard); then
+if ! $(true || usbDetectsRealtekCard); then
     echo -en "${yellow}[*] I wasn't able to find a Realtek card on your machine.  Do you want to proceed anyway? (Y/N)${restore}: "
     read PROCEED
 ```

I don't know if this driver will handle your card or not, but as long as you don't have any other Realtek devices on your system it won't hurt anything.

ieow, 2019-12-09:
Thanks.
I will check it out.