FreedomBen/rtl8188ce-linux-driver #145: Build failed from install.sh on 18.04

shawnmire, 2020-03-25:
Good afternoon!  Am on Ubuntu 18.04 and trying to install this driver but getting a build error.  I downloaded the zip from https://github.com/FreedomBen/rtl8188ce-linux-driver/tree/ubuntu-18.04 and ran the install.sh script.

Below is my terminal output from when it tried to compile:

<pre><font color="#3465A4">[*] Now let&apos;s compile the driver from source and copy the files to the right directories</font>
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
Yes
make -C /lib/modules/5.3.0-42-generic/build M=/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04 modules
make[1]: Entering directory &apos;/usr/src/linux-headers-5.3.0-42-generic&apos;
  CC [M]  /home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/base.o
In file included from <b>./arch/x86/include/asm/current.h:5:0</b>,
                 from <b>./include/linux/sched.h:12</b>,
                 from <b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/wifi.h:38</b>,
                 from <b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/base.c:33</b>:
<b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/base.c:</b> In function ‘<b>rtl_make_smps_action</b>’:
<b>./include/linux/compiler.h:78:22:</b> <font color="#AD7FA8"><b>warning: </b></font>this statement may fall through [<font color="#AD7FA8"><b>-Wimplicit-fallthrough=</b></font>]
 # define unlikely(x) <font color="#AD7FA8"><b>__builtin_expect(!!(x), 0)</b></font>
                      <font color="#AD7FA8"><b>^~~~~~~~~~~~~~~~~~~~~~~~~~</b></font>
<b>./include/asm-generic/bug.h:127:2:</b> <font color="#34E2E2"><b>note: </b></font>in expansion of macro ‘<b>unlikely</b>’
  <font color="#34E2E2"><b>unlikely</b></font>(__ret_warn_on);     \
  <font color="#34E2E2"><b>^~~~~~~~</b></font>
<b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/base.c:2234:3:</b> <font color="#34E2E2"><b>note: </b></font>in expansion of macro ‘<b>WARN_ON</b>’
   <font color="#34E2E2"><b>WARN_ON</b></font>( 1 );
   <font color="#34E2E2"><b>^~~~~~~</b></font>
<b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/base.c:2238:2:</b> <font color="#34E2E2"><b>note: </b></font>here
  <font color="#34E2E2"><b>case</b></font> IEEE80211_SMPS_OFF:/* 1 */ /*MIMO_PS_NOLIMIT*/
  <font color="#34E2E2"><b>^~~~</b></font>
  CC [M]  /home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/usb.o
  CC [M]  /home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/rc.o
<b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/rc.c:</b> In function ‘<b>rtl_get_rate</b>’:
<b>/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/rc.c:168:7:</b> <font color="#EF2929"><b>error: </b></font>implicit declaration of function ‘<b>rate_control_send_low</b>’; did you mean ‘<b>rate_control_set_rates</b>’? [<font color="#EF2929"><b>-Werror=implicit-function-declaration</b></font>]
  if ( <font color="#EF2929"><b>rate_control_send_low</b></font>( sta, priv_sta, txrc ) )
       <font color="#EF2929"><b>^~~~~~~~~~~~~~~~~~~~~</b></font>
       <font color="#4E9A06">rate_control_set_rates</font>
cc1: some warnings being treated as errors
scripts/Makefile.build:288: recipe for target &apos;/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/rc.o&apos; failed
make[2]: *** [/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04/rc.o] Error 1
Makefile:1655: recipe for target &apos;_module_/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04&apos; failed
make[1]: *** [_module_/home/shawn/Downloads/rtl8188ce-linux-driver-ubuntu-18.04] Error 2
make[1]: Leaving directory &apos;/usr/src/linux-headers-5.3.0-42-generic&apos;
Makefile:28: recipe for target &apos;all&apos; failed
make: *** [all] Error 2
<font color="#CC0000">[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)</font>
<font color="#8AE234"><b>shawn@endeavor</b></font>:<font color="#729FCF"><b>~/Downloads/rtl8188ce-linux-driver-ubuntu-18.04</b></font>$</pre>

seaniedan, 2020-07-14:
I get a similar result. I'm trying to get bluetooth working on a pavilion G6 with Realtek rtl8192ce:

sean@hp:~/rtl8188ce-linux-driver$ git checkout ubuntu-18.04
Already on 'ubuntu-18.04'
Your branch is up-to-date with 'origin/ubuntu-18.04'.
sean@hp:~/rtl8188ce-linux-driver$ make
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
You are running kernel 5.3.0-62-generic, which is not well supported.
See the README.md for recommended branch.
Continue with the build? (Y/N): y
make -C /lib/modules/5.3.0-62-generic/build M=/home/sean/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-62-generic'
  CC [M]  /home/sean/rtl8188ce-linux-driver/rc.o
/home/sean/rtl8188ce-linux-driver/rc.c: In function ‘rtl_get_rate’:
/home/sean/rtl8188ce-linux-driver/rc.c:168:7: error: implicit declaration of function ‘rate_control_send_low’; did you mean ‘rate_control_set_rates’? [-Werror=implicit-function-declaration]
  if ( rate_control_send_low( sta, priv_sta, txrc ) )
       ^~~~~~~~~~~~~~~~~~~~~
       rate_control_set_rates
/home/sean/rtl8188ce-linux-driver/rc.c: At top level:
/home/sean/rtl8188ce-linux-driver/rc.c:295:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .alloc = rtl_rate_alloc,
           ^~~~~~~~~~~~~~
/home/sean/rtl8188ce-linux-driver/rc.c:295:11: note: (near initialization for ‘rtl_rate_ops.alloc’)
cc1: some warnings being treated as errors
scripts/Makefile.build:288: recipe for target '/home/sean/rtl8188ce-linux-driver/rc.o' failed
make[2]: *** [/home/sean/rtl8188ce-linux-driver/rc.o] Error 1
Makefile:1664: recipe for target '_module_/home/sean/rtl8188ce-linux-driver' failed
make[1]: *** [_module_/home/sean/rtl8188ce-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-62-generic'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2

STPKITT, 2020-07-14:
I came across the same issue on a ThinkPad X121e running KDE neon (Ubuntu 18.04 under the hood):

`if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is generic-5.3.x based on your kernel version (5.3.0-62-generic)
Should I switch it to generic-5.3.x for you?  (y/n): y
Branch 'generic-5.3.x' set up to track remote branch 'generic-5.3.x' from 'origin'.
Switched to a new branch 'generic-5.3.x'
make -C /lib/modules/5.3.0-62-generic/build M=/home/stp/Treiber/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-62-generic'
  CC [M]  /home/stp/Treiber/rtl8188ce-linux-driver/base.o
  CC [M]  /home/stp/Treiber/rtl8188ce-linux-driver/usb.o
  CC [M]  /home/stp/Treiber/rtl8188ce-linux-driver/rc.o
/home/stp/Treiber/rtl8188ce-linux-driver/rc.c:300:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .alloc = rtl_rate_alloc,
           ^~~~~~~~~~~~~~
/home/stp/Treiber/rtl8188ce-linux-driver/rc.c:300:11: note: (near initialization for ‘rtl_rate_ops.alloc’)
cc1: some warnings being treated as errors
scripts/Makefile.build:288: recipe for target '/home/stp/Treiber/rtl8188ce-linux-driver/rc.o' failed
make[2]: *** [/home/stp/Treiber/rtl8188ce-linux-driver/rc.o] Error 1
Makefile:1664: recipe for target '_module_/home/stp/Treiber/rtl8188ce-linux-driver' failed
make[1]: *** [_module_/home/stp/Treiber/rtl8188ce-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-62-generic'
Makefile:28: recipe for target 'all' failed
make: *** [all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch) `

I hope this isn't the end of the Linux road for this otherwise still perfectly fine device.

FreedomBen, 2020-07-14:
@STPKITT and others, are you running distribution kernels or custom kernels of some sort?

It looks like the detection didn't notice you were on Ubuntu (maybe KDE Neon doesn't use the Canonical kernel) so might have picked the wrong branch.  You can try building ubuntu-18.04 (and tell the script no when it offers to switch you).  

If it builds it will most likely work, but there's always the risk of kernel panic so have a USB rescue image ready just in case

seaniedan, 2020-07-14:
I’m using Ubuntu mate 18.04. But even if I select that branch (see my previous message) it still fails. 

Sean

STPKITT, 2020-07-15:
Thanks for answering this issue!
I'm did nothing manually to change the kernel and AFAIK KDE neon just run whatever the latest Ubuntu LTS offers.
uname -r on that machine ouputs
> 5.3.0-62-generic
and when doing apt full-upgrade it more specifically says
> 5.3.0-62.56~18.04.1
Is that the issue, that being 18.04**.1** is means another branch is needed?
I already tried manual install, tried with branch set to 18.04 and 20.04 as well, but none of that worked.
Btw just using ./install.sh already mentions "You are running kernel 5.3.0.-62-generic, which is not well supported. See the README.md for recommended branch.". However in README.md I found no mention about recommended branch regarding that kernel version.
What I found strange on my machine as a non-expert was that when install.sh at the beginning tries to install dependencies it lists linux-headers-5.3.0.-62-generic and linux-headers-generic as already being the newest version though for the latter it's 4.15.0.111.99 which doesn't fit the 5.3.0.x pattern of the other component. Is that an issue?
Ok, I got nothing more to report right now but if I need to provide any more information or logs or so please let me know!

STPKITT, 2020-08-05:
I just wanted to let you guys know that I currently have it working again by manually installing kernel 5.4.15-050415-generic and switching the branch to that.

seaniedan, 2020-08-05:
Great! I’ll try it. Does Bluetooth work?

Sean

STPKITT, 2020-08-06:
I didn't try Bluetooth so you need to try it yourself and maybe report your findings here :-)

seaniedan, 2020-08-15:
Hi all, I managed to install the 5.4.15 kernel with 
wget -c https://kernel.ubuntu.com/~kernel-ppa/mainline//v5.4.15/linux-modules-5.4.15-050415-generic_5.4.15-050415.202001261031_amd64.deb 
wget -c https://kernel.ubuntu.com/~kernel-ppa/mainline//v5.4.15/linux-headers-5.4.15-050415_5.4.15-050415.202001261031_all.deb
...there were 2 more files on that webpage I had to get, then:
sudo dpkg -i *.deb
...to get that to compile I had to:
cd /lib/firmware/rtl_nic
sudo wget https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtl_nic/rtl8125a-3.fw
then cd back to where i'd saved the .deb files and
sudo dpkg -i *.deb
again... then I could run 
rm -rf rtl8188ce-linux-driver/ ; git clone https://github.com/FreedomBen/rtl8188ce-linux-driver.git && cd rtl8188ce-linux-driver && ./install.sh
...which compiled (on the second attempt - the first time I had an old repo which selected the wrong kernel version!)
Unfortunately, I get:
./am_i_using_this_driver.sh
[*] Not running the new rtlwifi (running stock driver)
[*] Not running the new rtl8192c_common (running stock driver)
[*] Not running the new rtl8192ce (running stock driver)
...and rebooting does not help. Any suggestions? Ben, do you know if this driver will enable Blutooth for this device? The card has shared bluetooth/wifi which works fine on Windows but not on  Linux.

lewisgit, 2021-01-06:
you can add -Wno-incompatible-pointer-types to extra flags in makefile to suppress the error.