FreedomBen/rtl8188ce-linux-driver #147: Constant faliure on PearOS (KDE)

the05animator, 2021-01-11:
No matter what i do (it even detects the system as Ubuntu 20.04) the build will always fail. may i have help?

FreedomBen, 2021-01-14:
Neat, first I've heard of PearOS.  Your distro might have custom kernel patches which break this build.  That's especially common with Ubuntu and Ubuntu derivatives.  Unfortunately I don't have any bandwidth to setup PearOS and work on that right now.

Your best bet is probably to go through the "Manual Installation" steps:  https://github.com/FreedomBen/rtl8188ce-linux-driver#manual-installation

When you get to where you select your branch, try a few different ones.  if `ubuntu-20.04` doesn't work, you can try the `generic-5.${m}.x` branch that corresponds to your kernel version (check with `uname -a`).  I would guess it's `generic-5.3.x` but you definitely want to verify.

If you get specific build failures, you can comment here with:

1.  Output of `uname -a` on your machine
2.  Which branch you tried to build
3.  The output from `make` where the failure occurs.  Best to include a dozen or so lines prior as well since the last line is not always the most useful line.

I or someone else may be able to give you more pointers at that point.  Good luck!

the05animator, 2021-01-21:
Ok thanks