FreedomBen/rtl8188ce-linux-driver #148: cant compile on T530 running Ubuntu 20.04

ivanbishop, 2021-01-11:
eading package lists... Done
Building dependency tree       
Reading state information... Done
gcc is already the newest version (4:9.3.0-1ubuntu2).
build-essential is already the newest version (12.8ubuntu1.1).
linux-headers-5.4.0-60-generic is already the newest version (5.4.0-60.67).
linux-headers-generic is already the newest version (5.4.0.60.63).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.25.1-1ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[*] Now let's compile the driver from source and copy the files to the right directories
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (current branch rtl8188ce-linux-driver-master)

You don't appear to be in a Git checkout.
This means branch information is not available.
You can still proceed to build, but you might be using unstable code.
Would you like me to try and get a git checkout for you? (Y/N): y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gcc is already the newest version (4:9.3.0-1ubuntu2).
build-essential is already the newest version (12.8ubuntu1.1).
linux-headers-5.4.0-60-generic is already the newest version (5.4.0-60.67).
linux-headers-generic is already the newest version (5.4.0.60.63).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.25.1-1ubuntu3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'rtl8188ce-linux-driver'...
remote: Enumerating objects: 104, done.
remote: Counting objects: 100% (104/104), done.
remote: Compressing objects: 100% (103/103), done.
remote: Total 8852 (delta 46), reused 3 (delta 1), pack-reused 8748
Receiving objects: 100% (8852/8852), 21.57 MiB | 753.00 KiB/s, done.
Resolving deltas: 100% (6830/6830), done.

Looks like the clone was successful (new folder is /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver)
Kicking off the build...

make[1]: Entering directory '/tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver'
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is ubuntu-20.04 based on your kernel version (5.4.0-60-generic)
Should I switch it to ubuntu-20.04 for you?  (y/n): y
Branch 'ubuntu-20.04' set up to track remote branch 'ubuntu-20.04' from 'origin'.
Switched to a new branch 'ubuntu-20.04'
make -C /lib/modules/5.4.0-60-generic/build M=/tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver modules
make[2]: Entering directory '/usr/src/linux-headers-5.4.0-60-generic'
  CC [M]  /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/base.o
  CC [M]  /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/usb.o
  CC [M]  /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/rc.o
/tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/rc.c:300:11: error: initialization of ‘void * (*)(struct ieee80211_hw *)’ from incompatible pointer type ‘void * (*)(struct ieee80211_hw *, struct dentry *)’ [-Werror=incompatible-pointer-types]
  300 |  .alloc = rtl_rate_alloc,
      |           ^~~~~~~~~~~~~~
/tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/rc.c:300:11: note: (near initialization for ‘rtl_rate_ops.alloc’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:275: /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver/rc.o] Error 1
make[2]: *** [Makefile:1757: /tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.4.0-60-generic'
make[1]: *** [Makefile:29: all] Error 2
make[1]: Leaving directory '/tmp/rtl8188ce-linux-driver-master/rtl8188ce-linux-driver'

Build exited with success status.
To install, cd over to rtl8188ce-linux-driver and run "sudo make install"
make -C /lib/modules/5.4.0-60-generic/build M=/tmp/rtl8188ce-linux-driver-master modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-60-generic'
scripts/Makefile.build:42: /tmp/rtl8188ce-linux-driver-master/Makefile: No such file or directory
make[2]: *** No rule to make target '/tmp/rtl8188ce-linux-driver-master/Makefile'.  Stop.
make[1]: *** [Makefile:1757: /tmp/rtl8188ce-linux-driver-master] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-60-generic'
make: *** [Makefile:29: all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)

FreedomBen, 2021-01-14:
Thanks, I'll check and see if upstream changed something that broke the 20.04 build.

FreedomBen, 2021-01-14:
@ivanbishop Can you pull the latest from ubuntu-20.04 and see if it's working now?  I was able to get it to build on my ubuntu machine by pushing https://github.com/FreedomBen/rtl8188ce-linux-driver/commit/f07579c29427f061b3b181b9de547290a2d5346b

ivanbishop, 2021-01-15:
Will tomorrow Ben, MANY thanks. I'll update the ticket/comments 
Ivan

phifgo, 2021-03-19:
I was unable  to compille 

Switched to a new branch 'generic-5.8.x'
make -C /lib/modules/5.8.0-45-generic/build M=/usr/src/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-45-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [Makefile:746: include/config/auto.conf] Error 1
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-45-generic'
make: *** [Makefile:29: all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)