FreedomBen/rtl8188ce-linux-driver #153: Failed to compile on Arch Linux

Redxx007, 2021-09-13:
I'm trying to compile on Arch Linux using ./install.sh and kernel 5.10-64-1-lts and I get this error:

/home/redxx/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c: In function ‘rtl88_hal_pwrseqcmdparsing’:
/home/redxx/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:70:17: error: implicit declaration of function ‘RT_TRACE’; did you mean ‘RTA_SPACE’? [-Werror=implicit-function-declaration]
 70 |                 RT_TRACE( rtlpriv, COMP_INIT, DBG_TRACE,
      |                 ^~~~~~~~
      |                 RTA_SPACE
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:280: /home/redxx/Downloads/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Erro 1
make[2]: *** [Makefile:1822: /home/redxx/Downloads/rtl8188ce-linux-driver/rtl8188ee] Erro 2
make[2]: Saindo do diretório '/usr/lib/modules/5.10.64-1-lts/build'

make[1]: *** [Makefile:33: default] Erro 2
make[1]: Saindo do diretório '/home/redxx/Downloads/rtl8188ce-linux-driver/rtl8188ee'

make: *** [Makefile:41: all] Erro 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)

project-administrator, 2021-09-15:
Same error on Debian

project-administrator, 2021-10-18:
The fix is here: https://github.com/project-administrator/rtl8192cu

PR #160 (2022-10-18): Fix compilation error ( #153, #155, #157, #159 ) - use rtl_dbg rather than RT_TRACE
This pull request fixes compilation errors caused by leftover calls to `RT_TRACE`, which has been obsoleted in favor of `rtl_dbg`.

Closes #153, closes #155, closes #157, closes #159.
closed 2022-10-18 by FreedomBen