FreedomBen/rtl8188ce-linux-driver #155: Compilation errors on Ubuntu 20.04 (current)

gsamborskipriv, 2021-12-22:
On the branch 
5.11.0-43-generic (default) 
I've have encountered the following compilation error:

pwrseqcmd.c: In function ‘rtl88_hal_pwrseqcmdparsing’:

pwrseqcmd.c:70:3: error: implicit declaration of function ‘RT_TRACE’; did you mean ‘RTA_SPACE’? [-Werror=implicit-function-declaration]
   70 |   RT_TRACE( rtlpriv, COMP_INIT, DBG_TRACE,
      |   ^~~~~~~~
      |   RTA_SPACE

On the branch ubuntu-20.04:  I've also have encountered the errors:

rtl8192cu/hw.c:622:2: warning: parameter names (without types) in function declaration
  622 |  u16 uninitialized_var( valuelow );

rtl8192cu/hw.c:627:3: error: ‘valuelow’ undeclared (first use in this function)
  627 |   valuelow = QUEUE_LOW;

Could You help  ?

ps. I've just purchased the network adapter Assus N15 which is titled as working on Linux but a driver for this card included in the kernel working  improperly . So I'm trying to use your driver.

FreedomBen, 2021-12-22:
@grzegorzspl are you using kernel 5.11?  If so is that a custom kernel?

FreedomBen, 2021-12-23:
@grzegorzspl  I did pick over some of the latest updates on ubuntu's 20.04 branch and push them up to the `ubuntu-20.04` branch.  I don't know if it will fix your compile issue, but it's worth a try.

If you are on a different kernel than the stock 5.4 though, it may be hard to build.  if you're on a vanilla mainstream kernel then you can usually use the corresponding branch, but if you're on a canonical maintained kernel they are sometimes hacked up enough that getting them to build is tough

grzegorzspl, 2021-12-25:
On Ubuntu 20.04 and the kernel 5.11: I commented out the lines with RT_TRACE then I managed to compile and It was working. I suppose that RT_TRACE means a macro referencing to the kernel sources.
----

The problem with RT_TRACE also exists on the current Fedora 35 which is my preferred Linux.

I commented it here (on Fedora 35) as well, compiled it, but failed to force Fedora to use this driver. Message below. I also tried to see the installation via dkms and manually. Without result.

How do I get Fedora to start using the compiled driver?

“
[*] I see you have an RTL8192CE card

[*] Now let's make sure your kernel loads the new modules at boot time

[*] Time to modprobe in your new driver...
[*] Not running the new rtlwifi (running stock driver)
[*] Not running the new rtl8192c_common (running stock driver)
[*] Not running the new rtl8192ce (running stock driver)
“

I cannot resolve the problem above by myself ? Any advise ?

------------------

ps. 
in regards to a custom Ubuntu kernel:

After reading your message that I’m using a custom kernel I changed it to the kernel 5.4 (GT) then I used the branch ubuntu-20.04. There, the installation was hassle-free. 
Ubuntu has a 5.11 kernel in HWE stack (Hardware Enablement) and 5.4 in GT stack. You're right that my kernel was from a ubuntu HWE stack, but I found out about it from you. That was a fresh ubuntu desktop installation.
https://wiki.ubuntu.com/Kernel/LTSEnablementStack

Thanks for Your answer.

grzegorzspl, 2021-12-27:
On Fedora 35 and kernel 5.15: I managed on my own. I found an advise to remove existing drivers: rm -rv /lib/modules/ uname -r/kernel/drivers/net/wireless/realtek/* then sudo ./install.sh and I installed this driver but I cannot reach tx-power:33 but 20 max on this driver on On Fedora 35 and kernel 5.15. It's strange, because on the same machine I have Ubuntu 20.04, kernel 5.4, where tx-power is 33

FreedomBen, 2021-12-27:
@grzegorzspl just checking, did you adjust the CRDA setting on the Fedora 35 machine?  To test it I usually set it to BO (Bolivia).  See 

> A very lenient CRDA domain is Bolivia (BO) that will let you up to 33 dBm

Under Troubleshooting step 3:  https://github.com/FreedomBen/rtl8188ce-linux-driver

le-est, 2022-01-01:
Can confirm that I run into the same error as @grzegorzspl on Ubuntu 20.04 and generic kernel 5.11.0.

marceloiury, 2022-05-19:
Hello Ben!

Thank you so much for this project. Would you please help me with a little doubt?

The README specifies ubuntu 22.04 as a supported version. Also, it is listed a branch for this version. However, i did not find this branch. When i ran the verify_branch, it is recommended  generic-5.1.x. It unfortunately does not build.  
Is there an ubuntu 22.04 branch?

In advance,  thank you so much for your support.

closed 2022-10-18 by FreedomBen

DreamShaded, 2022-10-19:
hello! 
compilation run not from master, but from generic-5.15.x

i've cherry-picked the changes and driver compiled successfully, but am_i_using script shows running stock driver =( 

what can i do with it?)

PR #160 (2022-10-18): Fix compilation error ( #153, #155, #157, #159 ) - use rtl_dbg rather than RT_TRACE
This pull request fixes compilation errors caused by leftover calls to `RT_TRACE`, which has been obsoleted in favor of `rtl_dbg`.

Closes #153, closes #155, closes #157, closes #159.