FreedomBen/rtl8188ce-linux-driver #157: Compilation Error on Generic 5.15.x

marceloiury, 2022-05-19:
Hello Fellows!

I got the following error when compiling the generic 5.15.x branch.  Any help?

/home/marcelo/Git/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:70:17: error: implicit declaration of function ‘RT_TRACE’; did you mean ‘RTA_SPACE’? [-Werror=implicit-function-declaration]
   70 |                 RT_TRACE( rtlpriv, COMP_INIT, DBG_TRACE,
      |                 ^~~~~~~~
      |                 RTA_SPACE
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:285: /home/marcelo/Git/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Error 1
make[2]: *** [Makefile:1875: /home/marcelo/Git/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-25-generic'
make[1]: *** [Makefile:33: default] Error 2
make[1]: Leaving directory '/home/marcelo/Git/rtl8188ce-linux-driver/rtl8188ee

FreedomBen, 2022-05-23:
@marceloiury what distro are you on?  and are you using the distro kernel or a custom kernel?

marceloiury, 2022-05-23:
I am using Ubuntu 22.04. Kernel 5.15. It is a distro kernel

VintageSteel, 2022-06-15:
- FWIW I'm also having exactly the same issue. I'm on Zorin OS 16.1, which is using 5.13.0-48-generic branch...

Dr-Pe, 2022-08-12:
Hi, I'm having the same error, using a freshly installed Ubuntu 22.04 which is using a 5.15.0-43-generic kernel.

iainism, 2022-08-17:
Same error here in Ubuntu Studio 22.04.1 (based on 5.15.0-46-lowlatency)

fran6120, 2022-09-15:
Same error here in `Kubuntu 22.04.1` using distro kernel `5.15.0-47-generic`

```
/home/javi/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:70:17: error: implicit declaration of function ‘RT_TRACE’; did you mean ‘RTA_SPACE’? [-Werror=implicit-function-declaration]
   70 |                 RT_TRACE( rtlpriv, COMP_INIT, DBG_TRACE,
      |                 ^~~~~~~~
      |                 RTA_SPACE
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:297: /home/javi/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Error 1
make[2]: *** [Makefile:1881: /home/javi/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-47-generic'
make[1]: *** [Makefile:33: default] Error 2
make[1]: Leaving directory '/home/javi/rtl8188ce-linux-driver/rtl8188ee'
make: *** [Makefile:41: all] Error 2
```

closed 2022-10-18 by FreedomBen

marceloiury, 2022-10-20:
> completed via #160

Thank you so much

automainint, 2022-12-01:
I had the same compilation error. Fixed it by replacing `RT_TRACE` with `rtl_dbg` in file `rtl8188ee/pwrseqcmd.c`

It seems, this issue still not fixed in branch `generic-3.15.x`

PR #160 (2022-10-18): Fix compilation error ( #153, #155, #157, #159 ) - use rtl_dbg rather than RT_TRACE
This pull request fixes compilation errors caused by leftover calls to `RT_TRACE`, which has been obsoleted in favor of `rtl_dbg`.

Closes #153, closes #155, closes #157, closes #159.