FreedomBen/rtl8188ce-linux-driver #158: Compile Error with generic-5.4.x branch on armv7l target

elockman, 2022-07-05:
I am trying to build a driver for an embedded linux board (Microchip, previously Atmel, SAMA5D27-XULT).  I am using the RTL8188 because is is affordable, available, and should be able to be configured as mesh point device.

lsusb show my hardware:
```
root@SAMA5D2:/# lsusb
Bus 001 Device 003: ID 0bda:f179 Realtek Semiconductor Corp. 802.11n
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

```
uname -a yields:
```
root@SAMA5D2:/# uname -a
Linux SAMA5D2 5.4.81-linux4sam-2020.10 #1 Thu Jan 14 12:54:56 UTC 2021 armv7l armv7l armv7l GNU/Linux

```
And the errors I get are:
```
root@SAMA5D2:/test/rtl8188ce-linux-driver# make
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
Yes
make -C /lib/modules/5.4.81-linux4sam-2020.10/build M=/test/rtl8188ce-linux-driver modules
make[1]: Entering directory '/lib/modules/5.4.81-linux4sam-2020.10/build'
  CC [M]  /test/rtl8188ce-linux-driver/base.o
/tmp/ccOJavb0.s: Assembler messages:
/tmp/ccOJavb0.s:1417: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccOJavb0.s:2270: Error: selected processor does not support `ldrex r2,[lr]' in ARM mode
/tmp/ccOJavb0.s:2272: Error: selected processor does not support `strex r4,r2,[lr]' in ARM mode
/tmp/ccOJavb0.s:2325: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccOJavb0.s:3459: Error: selected processor does not support `cpsid i' in ARM mode
/tmp/ccOJavb0.s:3886: Error: selected processor does not support `cpsid i' in ARM mode
make[2]: *** [scripts/Makefile.build:266: /test/rtl8188ce-linux-driver/base.o] Error 1
make[1]: *** [Makefile:1729: /test/rtl8188ce-linux-driver] Error 2
make[1]: Leaving directory '/lib/modules/5.4.81-linux4sam-2020.10/build'
make: *** [Makefile:29: all] Error 2
```
Any thoughts on how I can get around these?

FreedomBen, 2022-07-13:
Sorry elockman, that's a brutal one.  I've never tried building this on ARM. When I get a chance I'll try building it on my raspberry pi and see if I can work through the errors, but it won't be a quick fix (and depending on upstream support may not be fixable at all without serious effort).

elockman, 2022-07-13:
@FreedomBen thank you for the honest feedback.  I look forward to hearing what you find on your Rpi.

closed 2023-03-27 by elockman