FreedomBen/rtl8188ce-linux-driver #159: Suggestion to remove "RT_TRACE" because it is obsolete and has security problems

Ygor6889, 2022-07-16:
I suggest the creator of this driver remove everything related to "RT_TRACE" because this line is unnecessary and does nothing except give problems compiling the driver, this line is no longer needed because newer kernels don't need it anymore

FreedomBen, 2022-07-19:
Thanks @Ygor6889 !

trumad, 2022-09-25:
I just ran into a build failure because of this. Not quite sure how to fix it, but I'll try removing all "traces" of RT_TRACE

trumad, 2022-09-25:
Hooray, that worked. I ran the script from scratch, waited until the file rtl8188ee/pwrseqcmd.c existed, removed every call to RT_TRACE from it, and the script continued without issue. Unfortunately my wifi card still stops working intermittently, but that's a separate issue.

J0w03L, 2022-10-12:
> Hooray, that worked. I ran the script from scratch, waited until the file rtl8188ee/pwrseqcmd.c existed, removed every call to RT_TRACE from it, and the script continued without issue. Unfortunately my wifi card still stops working intermittently, but that's a separate issue.

Ran into the same problem and this worked for me too; thanks!

PR #160 (2022-10-18): Fix compilation error ( #153, #155, #157, #159 ) - use rtl_dbg rather than RT_TRACE
This pull request fixes compilation errors caused by leftover calls to `RT_TRACE`, which has been obsoleted in favor of `rtl_dbg`.

Closes #153, closes #155, closes #157, closes #159.
closed 2022-10-18 by FreedomBen