FreedomBen/rtl8188ce-linux-driver #165: The build has failed! received after upgrading to Linux Mint 21 (Vanessa)

jalsin82, 2023-04-16:
Below is the terminal history for trying to update the driver after updating Mint. Please help me get it to compile correctly.

p:~$ uname -a
Linux p 5.15.0-69-generic #76-Ubuntu SMP Fri Mar 17 17:19:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

p:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Linuxmint
Description:	Linux Mint 21
Release:	21
Codename:	vanessa

p:~$ lsudo apt-get install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.34.1-1ubuntu1.8).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

p:~$ lgit clone https://github.com/FreedomBen/rtl8188ce-linux-driver.git && cd rtl8188ce-linux-driver && ./install.sh
Cloning into 'rtl8188ce-linux-driver'...
remote: Enumerating objects: 9892, done.
remote: Counting objects: 100% (829/829), done.
remote: Compressing objects: 100% (216/216), done.
remote: Total 9892 (delta 632), reused 789 (delta 613), pack-reused 9063
Receiving objects: 100% (9892/9892), 21.71 MiB | 5.49 MiB/s, done.
Resolving deltas: 100% (7684/7684), done.

[*] So you want to live on the wild side and try a different driver for your RealTek wireless card eh?  Awesome!  I'll help you do it.

[*] We are going to build and install the driver from source code, compiled specifically for your machine.

[*] If you want to do the build/install manually, there are instructions in the "README.md" file.

[*] Please report any bugs/problems at https://github.com/FreedomBen/rtl8188ce-linux-driver

[*] You will need sudo privileges in order to complete this install.
[*] Press <Enter> when ready to begin, or <Ctrl+C> to quit

[*] Let's install any dependencies you will need in order to build the driver.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
gcc is already the newest version (4:11.2.0-1ubuntu1).
linux-headers-5.15.0-69-generic is already the newest version (5.15.0-69.76).
linux-headers-generic is already the newest version (5.15.0.69.67).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.34.1-1ubuntu1.8).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[*] Now let's compile the driver from source and copy the files to the right directories
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is generic-5.15.x based on your kernel version (5.15.0-69-generic)
Should I switch it to generic-5.15.x for you?  (y/n): y
Branch 'generic-5.15.x' set up to track remote branch 'generic-5.15.x' from 'origin'.
Switched to a new branch 'generic-5.15.x'
make -C /lib/modules/5.15.0-69-generic/build M=/home/jalsin82/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/base.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/usb.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rc.o
[... 111 lines cut ...]
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/hw.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/table.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/sw.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/trx.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/led.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/fw.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/phy.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/rf.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/dm.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/pwrseq.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o
/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c: In function ‘rtl88_hal_pwrseqcmdparsing’:
/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.c:70:17: error: implicit declaration of function ‘RT_TRACE’; did you mean ‘RTA_SPACE’? [-Werror=implicit-function-declaration]
   70 |                 RT_TRACE( rtlpriv, COMP_INIT, DBG_TRACE,
      |                 ^~~~~~~~
      |                 RTA_SPACE
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:297: /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee/pwrseqcmd.o] Error 1
make[2]: *** [Makefile:1906: /home/jalsin82/rtl8188ce-linux-driver/rtl8188ee] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-69-generic'
make[1]: *** [Makefile:33: default] Error 2
make[1]: Leaving directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee'
make: *** [Makefile:41: all] Error 2
[*] The build has failed! Please make sure you have all dependencies installed and you are building on the correct branch (kernel version, not linux distro version, is most important for the branch)

jalsin82, 2023-04-16:
Also tried the DKMS install with these results:

p:~$ sudo apt-get install git
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.34.1-1ubuntu1.8).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

p:~$ git clone https://github.com/FreedomBen/rtl8188ce-linux-driver.git
Cloning into 'rtl8188ce-linux-driver'...
remote: Enumerating objects: 9892, done.
remote: Counting objects: 100% (829/829), done.
remote: Compressing objects: 100% (216/216), done.
remote: Total 9892 (delta 632), reused 770 (delta 613), pack-reused 9063
Receiving objects: 100% (9892/9892), 21.71 MiB | 8.23 MiB/s, done.
Resolving deltas: 100% (7684/7684), done.

p:~$ sudo apt-get install gcc build-essential linux-headers-generic linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
gcc is already the newest version (4:11.2.0-1ubuntu1).
linux-headers-5.15.0-69-generic is already the newest version (5.15.0-69.76).
linux-headers-generic is already the newest version (5.15.0.69.67).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

p:~$ cd rtl8188ce-linux-driver && ./verify_branch.sh
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is generic-5.15.x based on your kernel version (5.15.0-69-generic)
Should I switch it to generic-5.15.x for you?  (y/n): y
Branch 'generic-5.15.x' set up to track remote branch 'generic-5.15.x' from 'origin'.
Switched to a new branch 'generic-5.15.x'

p:~/rtl8188ce-linux-driver$ sudo dkms install rtlwifi/1.0.0
Error! The /var/lib/dkms/rtlwifi/1.0.0/5.15.0-69-generic/x86_64/dkms.conf for module rtlwifi includes a BUILD_EXCLUSIVE directive which does not match this kernel/arch.
This indicates that it should not be built.

Ayush-Kumar-web, 2023-04-20:
Follow this guide bro

https://github.com/FreedomBen/rtl8188ce-linux-driver/pull/160

I encountered the same issue 
and solved it using this guide 
Contact me If you need any further assistance.

jalsin82, 2023-04-22:
I'm not very experienced with Linux. We just use it on an old computer 
for e-mail and internet.

I kind of understand the concept of replacing the old command with the 
new one. And, from what I can tell, there are 3 files that are impacted. 
Using Files (file browser) I searched for "rt_trace" and found the 
fix-rt-trace.sh file in /tools. But, in the original files downloaded 
using git, the 3 files didn't showed up.

I tried running install.sh, hoping that someone fixed it. But it failed 
with the attached fail log. I searched for "rt_trace" again and it found 
those 3 files this time. It seems old versions of these files are 
downloaded, or maybe "unpacked", as part of the install.? Regardless, 
after running the fix-rt-trace.sh tool on the new files and running the 
install.sh again, it was successful with the attached success log.

Hopefully, this output helps to fix it so that it runs successfully the 
first time through.

Thank you for your help!

Kind regards,

Cloning into 'rtl8188ce-linux-driver'...
remote: Enumerating objects: 9892, done.
remote: Counting objects: 100% (829/829), done.
remote: Compressing objects: 100% (216/216), done.
remote: Total 9892 (delta 632), reused 789 (delta 613), pack-reused 9063
Receiving objects: 100% (9892/9892), 21.71 MiB | 7.67 MiB/s, done.
Resolving deltas: 100% (7684/7684), done.

[*] So you want to live on the wild side and try a different driver for your RealTek wireless card eh?  Awesome!  I'll help you do it.

[*] We are going to build and install the driver from source code, compiled specifically for your machine.

[*] If you want to do the build/install manually, there are instructions in the "README.md" file.

[*] Please report any bugs/problems at https://github.com/FreedomBen/rtl8188ce-linux-driver

[*] You will need sudo privileges in order to complete this install.
[*] Press <Enter> when ready to begin, or <Ctrl+C> to quit

[*] Let's install any dependencies you will need in order to build the driver.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
gcc is already the newest version (4:11.2.0-1ubuntu1).
linux-headers-5.15.0-69-generic is already the newest version (5.15.0-69.76).
linux-headers-generic is already the newest version (5.15.0.70.68).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.34.1-1ubuntu1.8).
0 upgraded, 0 newly installed, 0 to remove and 14 not upgraded.

[*] Now let's compile the driver from source and copy the files to the right directories
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is generic-5.15.x based on your kernel version (5.15.0-69-generic)
Should I switch it to generic-5.15.x for you?  (y/n): y
Branch 'generic-5.15.x' set up to track remote branch 'generic-5.15.x' from 'origin'.
Switched to a new branch 'generic-5.15.x'
make -C /lib/modules/5.15.0-69-generic/build M=/home/jalsin82/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
[... 234 lines cut ...]
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-69-generic'
make[1]: Leaving directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8192cu'
make[1]: Entering directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8192se'
make -C /lib/modules/5.15.0-69-generic/build M=/home/jalsin82/rtl8188ce-linux-driver/rtl8192se modules
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-69-generic'
make[1]: Leaving directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8192se'
make[1]: Entering directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8192de'
make -C /lib/modules/5.15.0-69-generic/build M=/home/jalsin82/rtl8188ce-linux-driver/rtl8192de modules
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-69-generic'
make[1]: Leaving directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8192de'
make[1]: Entering directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee'
make -C /lib/modules/5.15.0-69-generic/build M=/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee modules
make[2]: Entering directory '/usr/src/linux-headers-5.15.0-69-generic'
make[2]: Leaving directory '/usr/src/linux-headers-5.15.0-69-generic'
make[1]: Leaving directory '/home/jalsin82/rtl8188ce-linux-driver/rtl8188ee'
if [ -e backup_existing.sh ] ; \
then \
    ./backup_existing.sh ; \
fi;
You already have a backup of a driver from this kernel version (located at /root/.rtlwifi-backup/5.15.0-69-generic.tar.gz.

You can back it up again, however if you have installed this driver already, then backing it up again will overwrite the original backup (which contains the stock drivers) with one that contains these drivers, which is most likely NOT what you want.  I recommend you only proceed with backing up the current drivers if you are sure of what you're doing.

jalsin82, 2023-04-22:
After getting it working, I was quickly notified there was another 
update to Mint. So, I updated to Vera (21.1). I ran the utility to check 
if I was using the new driver. Of course I wasn't. It took two attempts 
with running the fix utility in between again. I think they are the same 
outputs. But, in case they are different, both the failed first attempt 
and successful second attempt outputs are attached.

Kind regards,

Cloning into 'rtl8188ce-linux-driver'...
remote: Enumerating objects: 9892, done.
remote: Counting objects: 100% (829/829), done.
remote: Compressing objects: 100% (216/216), done.
remote: Total 9892 (delta 632), reused 770 (delta 613), pack-reused 9063
Receiving objects: 100% (9892/9892), 21.71 MiB | 6.61 MiB/s, done.
Resolving deltas: 100% (7684/7684), done.

[*] So you want to live on the wild side and try a different driver for your RealTek wireless card eh?  Awesome!  I'll help you do it.

[*] We are going to build and install the driver from source code, compiled specifically for your machine.

[*] If you want to do the build/install manually, there are instructions in the "README.md" file.

[*] Please report any bugs/problems at https://github.com/FreedomBen/rtl8188ce-linux-driver

[*] You will need sudo privileges in order to complete this install.
[*] Press <Enter> when ready to begin, or <Ctrl+C> to quit

[*] Let's install any dependencies you will need in order to build the driver.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
build-essential is already the newest version (12.9ubuntu3).
gcc is already the newest version (4:11.2.0-1ubuntu1).
linux-headers-5.15.0-70-generic is already the newest version (5.15.0-70.77).
linux-headers-5.15.0-70-generic set to manually installed.
linux-headers-generic is already the newest version (5.15.0.70.68).
The following package was automatically installed and is no longer required:
  mint-backgrounds-vanessa
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.34.1-1ubuntu1.8).
The following package was automatically installed and is no longer required:
  mint-backgrounds-vanessa
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

[*] Now let's compile the driver from source and copy the files to the right directories
if [ -e verify_branch.sh ] ; \
then \
    ./verify_branch.sh ; \
fi;
Verifying a sane branch for your kernel version...
No (Current branch master)
Recommended branch is generic-5.15.x based on your kernel version (5.15.0-70-generic)
Should I switch it to generic-5.15.x for you?  (y/n): y
Branch 'generic-5.15.x' set up to track remote branch 'generic-5.15.x' from 'origin'.
Switched to a new branch 'generic-5.15.x'
make -C /lib/modules/5.15.0-70-generic/build M=/home/jalsin82/rtl8188ce-linux-driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-70-generic'
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/base.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/usb.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/rc.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/debug.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/regd.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/efuse.o
  CC [M]  /home/jalsin82/rtl8188ce-linux-driver/cam.o
[... 256 lines cut ...]
find /lib/modules/5.15.0-70-generic -name "r8192ce_*.ko" -exec rm {} \;
find /lib/modules/5.15.0-70-generic -name "r8192cu_*.ko" -exec rm {} \;
find /lib/modules/5.15.0-70-generic -name "r8723e_*.ko" -exec rm {} \;
find /lib/modules/5.15.0-70-generic -name "r8188ee_*.ko" -exec rm {} \;
if [ -e update_modprobe_conf.sh ] ; \
then \
    ./update_modprobe_conf.sh ; \
fi;
Power save already disabled in /etc/modprobe.d/modprobe.conf. Doing nothing
[*] I see you have an RTL8192CE card

[*] Now let's make sure your kernel loads the new modules at boot time

[*] Time to modprobe in your new driver...
modprobe: FATAL: Module rtlwifi is in use.
[*] You are running the new rtlwifi
[*] You are running the new rtl8192c_common
[*] You are running the new rtl8192ce

[*] If you aren't running the driver, try a reboot and re-run the script "am_i_using_this_driver.sh".

[*] NOTE: If your wifi looks dead, it usually comes back after a reboot.  Don't panic yet.

[*] OK, ready for a reboot. (only necessary if your wifi is not working)
[*] You can check if you're running the new driver by running the script "am_i_using_this_driver.sh"

Ayush-Kumar-web, 2023-10-14:
sorry for such a long late reply 
What is the condition now is it wokring?
@jalsin82

thefangeddeity, 2023-10-19:
> What is the condition now is it working?

Works great. Thank you!

Sent from [Proton Mail](https://proton.me/mail/home) for iOS